espressif/esp-idf #1381: BT: Memory leak when using encrypted connections

rojer, 2017-12-13:
Enabling encryption on a  BT connection leaks memory on every connection.
If we add printing free heap to `examples/bluetooth/gatt_server` and `examples/bluetooth/`

Yulong-espressif, 2017-12-16:
@rojer I think we have fix this issue in the new version of SDK. And I has test it in my version(8f874aaa), there is no memery leak now. Thank you very much.

Yulong-espressif, 2017-12-16:
I (48056) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT                                                                    
I (49886) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LENC                                                               
I (49886) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LID                                                                
I (50076) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PENC
I (50076) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PID
I (50076) SEC_GATTS_DEMO: remote BD_ADDR: 46ef75686dc3
I (50076) SEC_GATTS_DEMO: address type = 1
I (50086) SEC_GATTS_DEMO: pair status = success
I (50096) SEC_GATTS_DEMO: Bonded devices number : 2

I (50096) SEC_GATTS_DEMO: Bonded devices list : 2

I (50106) SEC_GATTS_DEMO: 46 ef 75 68 6d c3 
I (50106) SEC_GATTS_DEMO: 74 b7 b1 4a ff 40 
I (60806) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, heap_size = 152388
I (60806) SEC_GATTS_DEMO: advertising start success
I (66606) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (80556) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, heap_size = 152388
I (80556) SEC_GATTS_DEMO: advertising start success
I (101876) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (112626) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, heap_size = 152388
I (112626) SEC_GATTS_DEMO: advertising start success
I (116926) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (138416) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, heap_size = 152388
I (138416) SEC_GATTS_DEMO: advertising start success
I (145636) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (151516) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, heap_size = 152388
I (151516) SEC_GATTS_DEMO: advertising start success
I (155506) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (162106) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, heap_size = 152388
I (162116) SEC_GATTS_DEMO: advertising start success

rojer, 2017-12-18:
synced to latest master and still seeing the same behavior:
```diff
$ git diff
diff --git a/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c b/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c
index eb328c4..9dd9850 100644
--- a/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c
+++ b/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c
@@ -394,7 +394,7 @@ static void gatts_profile_event_handler(esp_gatts_cb_event_t event,
         case ESP_GATTS_STOP_EVT:
             break;
         case ESP_GATTS_CONNECT_EVT:
-            ESP_LOGI(GATTS_TABLE_TAG, "ESP_GATTS_CONNECT_EVT");
+            ESP_LOGI(GATTS_TABLE_TAG, "ESP_GATTS_CONNECT_EVT %d", xPortGetFreeHeapSize());
             /* start security connect with peer device when receive the connect event sent by the master */
             esp_ble_set_encryption(param->connect.remote_bda, ESP_BLE_SEC_ENCRYPT_MITM);
             break;
```

```
[Dec 18 11:56:30.613] I (104) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
[Dec 18 11:56:30.882] I (374) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0
[Dec 18 11:56:30.882] I (374) SEC_GATTS_DEMO: app_main init bluetooth
[Dec 18 11:56:30.924] I (424) SEC_GATTS_DEMO: The number handle = 8
[Dec 18 11:56:30.932] I (424) SEC_GATTS_DEMO: advertising start success
[Dec 18 11:56:44.361] I (13854) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT 168608
[Dec 18 11:56:44.829] I (14304) SEC_GATTS_DEMO: remote BD_ADDR: 7c66d4970b2b
[Dec 18 11:56:44.829] I (14304) SEC_GATTS_DEMO: address type = 1
[Dec 18 11:56:44.829] I (14304) SEC_GATTS_DEMO: pair status = success
[Dec 18 11:56:44.829] I (14314) SEC_GATTS_DEMO: Bonded devices number : 1
[Dec 18 11:56:44.829] 
[Dec 18 11:56:44.829] I (14314) SEC_GATTS_DEMO: Bonded devices list : 1
[Dec 18 11:56:44.838] 
[Dec 18 11:56:44.838] I (14324) SEC_GATTS_DEMO: 7c 66 d4 97 0b 2b 
[Dec 18 11:56:49.903] I (19394) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
[Dec 18 11:56:49.903] I (19394) SEC_GATTS_DEMO: advertising start success
[Dec 18 11:56:52.282] I (21774) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT 168132
[Dec 18 11:56:52.749] I (22224) SEC_GATTS_DEMO: remote BD_ADDR: 7c66d4970b2b
[Dec 18 11:56:52.750] I (22224) SEC_GATTS_DEMO: address type = 1
[Dec 18 11:56:52.750] I (22224) SEC_GATTS_DEMO: pair status = success
[Dec 18 11:56:52.750] I (22234) SEC_GATTS_DEMO: Bonded devices number : 1
[Dec 18 11:56:52.750] 
[Dec 18 11:56:52.750] I (22234) SEC_GATTS_DEMO: Bonded devices list : 1
[Dec 18 11:56:52.759] 
[Dec 18 11:56:52.759] I (22244) SEC_GATTS_DEMO: 7c 66 d4 97 0b 2b 
[Dec 18 11:56:57.058] I (26544) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
[Dec 18 11:56:57.058] I (26544) SEC_GATTS_DEMO: advertising start success
[Dec 18 11:56:58.133] I (27624) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT 167820
[Dec 18 11:56:58.645] I (28124) SEC_GATTS_DEMO: remote BD_ADDR: 7c66d4970b2b
[Dec 18 11:56:58.646] I (28124) SEC_GATTS_DEMO: address type = 1
[Dec 18 11:56:58.646] I (28124) SEC_GATTS_DEMO: pair status = success
[Dec 18 11:56:58.646] I (28124) SEC_GATTS_DEMO: Bonded devices number : 1
[Dec 18 11:56:58.646] 
[Dec 18 11:56:58.646] I (28134) SEC_GATTS_DEMO: Bonded devices list : 1
[Dec 18 11:56:58.655] 
[Dec 18 11:56:58.655] I (28134) SEC_GATTS_DEMO: 7c 66 d4 97 0b 2b 
[Dec 18 11:57:04.439] I (33924) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
[Dec 18 11:57:04.439] I (33924) SEC_GATTS_DEMO: advertising start success
[Dec 18 11:57:04.705] I (34194) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT 167512
[Dec 18 11:57:05.217] I (34694) SEC_GATTS_DEMO: remote BD_ADDR: 7c66d4970b2b
[Dec 18 11:57:05.217] I (34694) SEC_GATTS_DEMO: address type = 1
[Dec 18 11:57:05.217] I (34694) SEC_GATTS_DEMO: pair status = success
[Dec 18 11:57:05.217] I (34694) SEC_GATTS_DEMO: Bonded devices number : 1
[Dec 18 11:57:05.217] 
[Dec 18 11:57:05.217] I (34704) SEC_GATTS_DEMO: Bonded devices list : 1
[Dec 18 11:57:05.226] 
[Dec 18 11:57:05.226] I (34704) SEC_GATTS_DEMO: 7c 66 d4 97 0b 2b 
```

Yulong-espressif, 2017-12-18:
@rojer Ok, thank you very much. We will test again and response to you as soon as posible.

Yulong-espressif, 2017-12-19:
@rojer We have find the issues now, we will update it to the new version of SDK soon. Thanks again.

FayeY, 2018-01-02:
Hi @rojer , this issue should have been solved in the latest IDF, please have a try, and feel free to reopen if it is still unsolved. Thanks.

closed 2018-01-02 by FayeY

rojer, 2018-01-02:
tested, can confirm that the bug has been fixed. thanks!