espressif/esp-idf #1383: ble_ibeacon IBEACON_SENDER (ESP-IDF Pre-release 3.0-rc1)

Daivyge, 2017-12-14:
HI all;
         Use ble_ibeacon demo code in example (IBEACON_SENDER)  I try to send ibeacon adv. 
        After burning and reboot i can receive the device adv with my iphone, but adv data is empty.
        if I change the company id,then I can see the adv data ,but i know this isn't iBeacon.

TianHao-Espressif, 2017-12-21:
@Daivyge , Please do not use nRF connect or LightBlue in IOS. I have tried it, the application in IOS cannot recognize the IBeacon. But  nRF connect on Android can recognize it. I have tried with ESP32 IBeacon demo and other device named RadBeacon. I don't know why the application in IOS cannot recognize it, I guess that the company doesn't register IBeacon service with Apple.

Daivyge, 2017-12-27:
@TianHao-Espressif ,
OK, I get it.
Thanks very much.

closed 2017-12-28 by Daivyge