espressif/esp-idf #1385: Http request crashes application

llewellynnu, 2017-12-14:
Hi, 

In my application I am using UART1 - GSM(PPP), UART2 - GPS, Timer, AWSIoT, BTLE 
The application is running on a custom board with ESP32-WROVER

When I make an HTTP Request the application crashes with the following error. 

assertion "fd >=0 && fd < 3" failed: file "/Users/souravjena/esp/esp-idf/components/vfs/./vfs_uart.c", line 187, function: uart_read
abort() was called at PC 0x400f44cf on core 1
0x400f44cf: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x4008b8a0:0x3ffe1080 0x4008b99f:0x3ffe10a0 0x400f44cf:0x3ffe10c0 0x40107be6:0x3ffe10f0 0x401077e1:0x3ffe1110 0x4000bdbb:0x3ffe1130 0x400017ea:0x3ffe1150 0x400ed3cb:0x3ffe1170 0x400ed706:0x3ffe12a0
0x4008b8a0: invoke_abort at /Users/souravjena/esp/esp-idf/components/esp32/./panic.c:572

0x4008b99f: abort at /Users/souravjena/esp/esp-idf/components/esp32/./panic.c:572

0x400f44cf: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x40107be6: uart_read at /Users/souravjena/esp/esp-idf/components/vfs/./vfs_uart.c:117

0x401077e1: esp_vfs_read at /Users/souravjena/esp/esp-idf/components/vfs/./vfs.c:506 (discriminator 4)

0x400ed3cb: GetUpdateFile at /Users/souravjena/esp/nu-m2-firmware/main/./otaM2.c:352

0x400ed706: OTATask at /Users/souravjena/esp/nu-m2-firmware/main/./otaM2.c:96

What could be the issue?

line 352 on otaM2.c is ret = read(s, recv_buf, sizeof(recv_buf) - 1);

Regards

llewellynnu, 2017-12-20:
@igrr Thoughts?

igrr, 2017-12-20:
It looks like `s` should be a socket number, i.e. LWIP_SOCKET_OFFSET + something, but it ends up being something smaller that LWIP_SOCKET_OFFSET, i.e. a value near zero.
Can you print out `s` value before calling `read` and at the point when you have opened a socket.

closed 2017-12-29 by llewellynnu