espressif/esp-idf #1389: I2C read error

sjoteppagol, 2017-12-15:
We have used GPIO 25 and GPIO 4 as SDA and SCL respectively. Tested separately and it's working.
We have used GPIO 21 as interrupt pin and GPIO 22 as normal GPO(LED). Tested separately and it's working. But when embedded both code into single application i am getting I2C read error.

Is there any particular bit set by default in I2C, which says GPIO 21 and 22 are I2C pins as I2C by default. Even though i am using GPIO 21 as interrupt pin(push button) and GPIO 22 as normal GPIO(LED)?

Spritetm, 2017-12-15:
There is no default pin for the I2C peripheral. Can you post your code somewhere so we can see if it's a coding issue and/or reproduce your error?

FayeY, 2017-12-19:
Hi @sjoteppagol , is there any update about this issue? Could you provide your source code so we can help debug it? Thanks.

sjoteppagol, 2017-12-19:
The issue is solved. I tried to do pin configuration with bit shift, this solved the issue. 
**Before:**
` io_conf.pin_bit_mask = pin;`

  **After:**
`io_conf.pin_bit_mask |= (1 << temp[loopCount].pin);`

Thank you for your quick response.

costaud, 2017-12-19:
@sjoteppagol  Note that it would be better to use `(1ULL << temp[loopCount].pin)`, in case the pin index is bigger than 31.

closed 2017-12-19 by FayeY