espressif/esp-idf #1391: esp_timer studdenly pause

v1993, 2017-12-15:
I'm developing module which play audio via DAC. To do it, I use `esp_timer` interface. But sound may stop (it simply doesn't run callback). If another task will produce watchdog restart few times, it will restart after some time. After restarting timer by `esp_timer_start_periodic` everything is fine again.

This bug is critical for `esp_timer` module, because it mean that timer does not always run.

P.S. It appears on 8000Hz and 16000Hz, this two is confirmed.

Spritetm, 2017-12-15:
Not to detract from your bug report, but are you aware that the I2S peripheral can fully automatically control the DACs as well? In my experience, it's much more suited to audio generation.

v1993, 2017-12-15:
Thank you for advice, but I'm still think that it is actual bug.

Spritetm, 2017-12-15:
Ah yes, as I said, the option to also ise the I2S driver does not detract from the fact that there's still a bug here, and that that needs to be solved. Just wanted you to know that if your goal is to generate audio from the DACs, there's also a less painful way.

v1993, 2017-12-15:
I looked at I2S. It looks good, but doesn't apporative for me, because I want to play more tan one track at same time (via software mixing).

Just found that it may restart without wathcdog triggering. For me, it looks like overflow.

v1993, 2018-01-05:
Is there any progress on this issue? Can I help somehow?

igrr, 2018-01-05:
There was a recent fix for esp_timer in https://github.com/espressif/esp-idf/commit/0a1f81dc01f4e3a663b1c2b309e457cd11703ced, is the commit you are testing more recent than that one?

v1993, 2018-01-05:
Thank you. I'll reopen this issue if catch it again.

closed 2018-01-05 by v1993