espressif/esp-idf #1392: No dac output in a2dp sink example

Ironman2696, 2017-12-15:
I have configured DAC as output for a2dp sink example, but the there is no output on gpio 25 and 26. And there is no dac.c file in driver folder, should i be worried?

tuanpmt, 2017-12-15:
Please try to add `i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN);` after https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/a2dp_sink/main/main.c#L73

Ironman2696, 2017-12-16:
Thank you tuanpmt, it dint work.  i am building on windows, does this cause the problem?

pranay96, 2017-12-16:
i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN); is called from i2s_set_pin(0, NULL);  so no need to add it in main.c . i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN); calls a method dac_i2s_enable(); to enable both the DACS . dac_12s_enable() is defined in dac.h, but as you said there is no dac.c file, so that must be the problem.

closed 2017-12-21 by FayeY