espressif/esp-idf #1395: Uart is slow

bookerdong, 2017-12-16:
Hello,
        I was using the UART asynchronous example project found in my bus from the detected data to I receive data, after about 400us, I feel this time a bit too high. I use the baud rate is 230400, how can we shorten the time, because we are the time for us,
Thank you

bookerdong, 2017-12-28:
I found esp32 no non-interrupted serial port

koobest, 2018-01-03:
Hi @bookerdong, When the data comes, the uart interrupt processing flow are as follows:
fifo -> buffer-> ring buffer -> your buffer
so it may taks a long time to read the data. so I think the one solution is to re-write the interrupt service handler  based on your application.

koobest, 2018-01-03:
When the data come, you read the data to you buffer from uart fifo.

FayeY, 2018-01-22:
Closing due to lack of response, please feel free to reopen if this is still an issue.

closed 2018-01-22 by FayeY

snahmad, 2018-07-28:
did you manged to solve UART slow issue. I have my circular buffer. I need to configure uart 0 or 1 or 2 and use my own interrupt handler. Do any one have any working example code?