espressif/esp-idf #1396: nvs_flash_init issue

hvishu, 2017-12-16:
HI,
I am facing random crash when using nvs_flash_init().
nvs_flash_init(), crashes randomly, and by having a printf (with different length), the problem goes away.  But again changes in some other part of the code causes the same issue.  Again need to use additional printfs.  I have attached dumps of working and not working cases.  As you can see from them "nvs_not_working_case" crashes and it gets corrected in "nvs_working_case" by having additional printf "1111111111111111111223".
I have prints in nvs_flash_init and function it calls.  From the log, you can observe that crash happens before nvs_flash_init_custom (line number 100).  I have attached the nvs_api.cpp file which contains the print statements.   
As shown in log files the git tag I am using is "09d2791cfdf7"
Please help.  Let me know if you need any further information.
[nvs_dump.zip](https://github.com/espressif/esp-idf/files/1564981/nvs_dump.zip)
[nvs_api.cpp.zip](https://github.com/espressif/esp-idf/files/1564980/nvs_api.cpp.zip)

Thanks and Regards,
Vishwa

igrr, 2017-12-16:
Your log output contains messages
```
Timer ISR init is done
```
and 
```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
```
which suggests that you might be using Timer Group driver with an ISR which has `ESP_INTR_FLAG_IRAM` flag, and the ISR is calling a function (or accessing constant data) which is in IRAM.

If you use idf_monitor tool (`make monitor`), it will decode the backtrace output from the exception and point you to the place where the issue happens.

hvishu, 2017-12-17:
Thank you,
Solved my problem.  
As you mentioned the timer ISR was indeed using the flag ESP_INTR_FLAG_IRAM and ISR was accessing a function (found using make monitor).  By not calling it solved the problem.
But couldn't understand the relation,  between the flag ESP_INTR_FLAG_IRAM (ISR can be called if cache is disabled) and the called function which is declared as bool IsWifiModulePresent(void).   From map file for the same function below references are found
 .literal.IsWifiModulePresent
                0x0000000040108b44        0x0     
 .text.IsWifiModulePresent
                0x000000004010f570       0x14
Could you provide some more information on what exactly is the reason for crash (so that we can avoid this in future)
Regards,
Vishwa

igrr, 2017-12-17:
ISR may get called when cache is disabled, and the function you are calling from ISR is not in IRAM (4010f570 is cache region), so CPU can not fetch the instruction when cache is disabled.

Please see http://esp-idf.readthedocs.io/en/latest/api-reference/system/intr_alloc.html#iram-safe-interrupt-handlers and http://esp-idf.readthedocs.io/en/latest/api-guides/general-notes.html#iram-instruction-ram.

closed 2017-12-19 by FayeY