espressif/esp-idf #1398: Description	Resource	Error: make: *** [/home/Vima/esp-idf/components/esptool_py/Makefile.projbuild:55: flash] Error 1

mkvn, 2017-12-17:


mkvn, 2017-12-17:
Hi, 
I try to bild my project for Esp32 development board. When I Build my project (before doing make flash) in eclipse (inside windows), I see this error in tab Problems of eclipse:

make: *** [/home/Vima/esp-idf/components/esptool_py/Makefile.projbuild:55: flash] Error 1	
mentioned resource is : esp-idf-template		 	

I have googled it but couldn't undrestand how to solve it, I appreciate any solution or suggestion.

projectgus, 2017-12-17:
Can you please give us a few more details so we can help resolve this problem:

* What OS are you using?
* Did you follow all of the [steps in the Eclipse setup guide](https://esp-idf.readthedocs.io/en/latest/get-started/eclipse-setup.html)?
* What is the full log output in the Eclipse Console tab?

mkvn, 2017-12-18:
Hi,
My OS is Windows 8.1. 
and yes I have followed all the steps in the Eclipse setup guide and I managed to run one program before. 
my full log out put in Eclipse Console is very long . so I attached it in a seperate file.
Thank you for your help , I'm looking forward your answer.
Cheers

[FullConsole.txt](https://github.com/espressif/esp-idf/files/1567435/FullConsole.txt)

mkvn, 2017-12-18:
Hi again Projectgus,
I guess I have sent the wrong cosole to you, the attached file here is related to console when I do flash on my program i2c_Test. I hope you can find a soluion for it.
[i2cTest_Console.txt](https://github.com/espressif/esp-idf/files/1568946/i2cTest_Console.txt)

projectgus, 2017-12-19:
If I search the output of the Console for Error then I see this:
```
serial.serialutil.SerialException: could not open port 'COM3': WindowsError(5, '\xc5tkomst nekad.')
```

I'm not sure why Eclipse is parsing this incorrectly, but it seems like this is the error.

There are two possibly ways you can fix this:

* Go into Properties of your project, Environment Variables, and set ESPPORT to the correct COM port (ie COMx).
* OR run "make menuconfig" on the command line and set the COM port in the SDK config under "Serial Flasher Config".

The first option is probably easier.

PS For future reference, a great way to post log files is to paste into https://pastebin.com or https://gist.github.com and then post a link.

mkvn, 2017-12-19:
Thank you Projectgus for your suggestions.
I had already run "make menuconfig", but it didn't solve the problem. Now I used yor first suggestion and it helped.

khoa21214475, 2018-02-10:
me too...need helps pls!
esptool.py v2.2.1
Flashing binaries to serial port /dev/ttyUSB3 (app at offset 0x10000)...
esptool.py v2.2.1
Traceback (most recent call last):
  File "C:/msys32/home/Admin/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2599, in <module>
    _main()
  File "C:/msys32/home/Admin/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2592, in _main
    main()
  File "C:/msys32/home/Admin/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2319, in main
    esp = chip_class(args.port, initial_baud, args.trace)
  File "C:/msys32/home/Admin/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 187, in __init__
    self._port = serial.serial_for_url(port)
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'C:/msys32/dev/ttyUSB3': WindowsError(2, 'The system cannot find the file specified.')
make: *** [/home/Admin/esp/esp-idf/components/esptool_py/Makefile.projbuild:55: flash] Error 1

negativekelvin, 2018-02-10:
@khoa21214475 use Windows port names

khoa21214475, 2018-02-10:
@negativekelvin : 

I changed port name from 0--3, but i got the same results, :(

negativekelvin, 2018-02-10:
@khoa21214475 windows port name uses COM not /dev/ttyUSB

khoa21214475, 2018-02-10:
@negativekelvin:
you mean i change /dev/ttyUSB0 to COM0, right? I will try

negativekelvin, 2018-02-10:
It starts with COM1, look in device manager

khoa21214475, 2018-02-10:
ok, i see, im trying to test again with COM3 ( look in Device manager). Thanks! :)

closed 2019-03-01 by projectgus