espressif/esp-idf #1401: Enhancement: Query and return the list of white listed addresses

nkolban, 2017-12-18:
We have APIs to add a BLE address and remove a BLE address from the BLE white list ... however we don't seem to have an API to query the list of BLE addresses on the white list.  This issue asks for the consideration of an additional API to return the white list set.

blueMoods, 2017-12-25:
Bluedroid does not provide this command, so we do not intend to increase this API, thank you for your advice.
/home/baohongde/Desktop/Screenshot from 2017-12-25 10-47-40.png

blueMoods, 2017-12-25:
Bluedroid does not provide this command, so we do not intend to increase this API, thank you for your advice.

blueMoods, 2017-12-25:
[image]

nkolban, 2017-12-26:
Thank you sir.  I guess if the underlying environment doesn't provide it, that's the end of the story.

closed 2017-12-26 by nkolban