espressif/esp-idf #1404: esp_wifi_set_promiscuous_filter set WIFI_PROMIS_FILTER_MASK_ALL error

malaimoo, 2017-12-18:
When set promiscuous mode true, and set the filter to all, it comes to error 1024 sometimes.
```
    wifi_promiscuous_filter_t filter = {
        .filter_mask = WIFI_PROMIS_FILTER_MASK_ALL
        };
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter(&filter));
```
and the error

```
 wdev.c wdevProcessRxSucDataAll 1024
```

by the way, I want use esp32 to sniffer CTS frame , the control byte like "01 0c"  Is it possible?

malaimoo, 2017-12-27:
@FayeY Do you have some suggestions? Hope to get your response..

malaimoo, 2017-12-28:
@igrr Hello , Sorry to bother you, but I am desperate to solve this problem, can you give me some suggestion?

TimXia, 2017-12-28:
@malaimoo Please set the filter before setting promiscuous mode true. ESP32 can not capture control frame like RTS, CTS and ACK.

malaimoo, 2017-12-28:
@TimXia Thanks for your  response ，but I did set set the filter before setting promiscuous mode true, My code is like 

```
        ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
        wifi_promiscuous_filter_t filter = {
		.filter_mask = WIFI_PROMIS_FILTER_MASK_ALL};
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter(&filter));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous_rx_cb(wifi_promiscuous_cb));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true));
```
As I know , Control Frame is necessary in 802.11 communication, Does IDF just drop control frame or the esp32 hardware did not have the ability to capture control frame ?

TimXia, 2017-12-28:
@malaimoo IDF drops control frame. We may consider capture it later. Can you provide a demo to reproduce the error?

TimXia, 2017-12-28:
@malaimoo And also provide IDF version and commit ID please.

malaimoo, 2017-12-28:
@TimXia 
Thank you for your prompt reply. 

IDF Version 3.0 rc1
commit ID :043ef32
add the demo:
[esp32_sniffer.zip](https://github.com/espressif/esp-idf/files/1590604/esp32_sniffer.zip)

Because our production need to capture CTS , and without this function our production can't be finished. So, if possible ,I'm very much looking forward to see this function in the future version. 
Thank a lot!

TimXia, 2017-12-28:
@malaimoo OK. I will use the demo to reproduce the error and try to find the root cause.

liuzfesp, 2018-01-22:
@malaimoo the root cause has been found, @TimXia will update the fixed into v3.1/v3.0

TimXia, 2018-01-24:
@malaimoo The issue has been solved in IDFv3.0. Please pull the latest IDFv3.0 to test. Thanks,

closed 2018-01-26 by FayeY