espressif/esp-idf #1405: BT: How to find out local BT MAC?

rojer, 2017-12-18:
Maybe i'm blind, but i don't see a function to get the address used by BT controller.
we have `esp_ble_gap_set_rand_addr` to set random address, but i'd rather use `esp_ble_gap_config_local_privacy` to have UDF generate it for me. the problem is, there doesn't seem to be a way to find out the value after it's generated.

nkolban, 2017-12-18:
Howdy my friend.  Might the function called `esp_bt_dev_get_address()` be what you are looking for?

See:

http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_bt_device.html?highlight=esp_bt_dev_get_address#_CPPv222esp_bt_dev_get_addressv

FayeY, 2017-12-21:
Hi @rojer , did the above suggestion help you resolve the issue?

rojer, 2017-12-21:
yes, thanks!

rojer, 2018-01-02:
it seems to always return the public address (`24:0a:...`), even if random address is enabled (`esp_ble_gap_config_local_privacy(true)`).

chegewara, 2018-01-02:
Check this esp-idf example. I think you need to set advertising param flag to BLE_ADDR_TYPE_RANDOM. This example for sure setup randomized MAC

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c#L90

rojer, 2018-01-02:
@chegewara yes, i set addr type to random in the adv params. in general, i have no trouble setting up the address - it's random, and clients can connect, no problem. but i need to find out what the address _is_, once it's set up.

negativekelvin, 2018-01-03:
Random address is updated based on timer so you also would have to be aware of timer when showing user what the "current" random address is.

chegewara, 2018-01-03:
@rojer I did not check it but maybe try to read local mac with esp_bt_dev_get_address() during this event:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c#L338

rojer, 2018-01-03:
@negativekelvin still, it would be nice to find out what the current address is.
@chegewara that's exactly what i did and it still returned the public address.

TianHao-Espressif, 2018-01-05:
esp_bt_dev_get_address() always return the real(public) address. Or you want to get the random address which is in used, right?

rojer, 2018-01-05:
@TianHao-Espressif yes

Weijian-Espressif, 2018-03-13:
@rojer sorry there is no such interface yet.

Weijian-Espressif, 2018-04-02:
@rojer We have added this API, and we are testing it now. We will merge to master later.

rojer, 2018-04-02:
@Weijian-Espressif thank you!

FayeY, 2018-04-08:
It has been merged, please have a try. Thanks.

closed 2018-04-08 by FayeY

rojer, 2018-06-19:
finally got to try this. the API is `esp_ble_gap_get_local_used_addr()` and it works ok. thank you!