espressif/esp-idf #1406: BT: Provide a way to reset BT settings

rojer, 2017-12-18:
Including bonded devices and whatever else is stored.
This is necessary for "factory reset" function. Currently, when factory reset is requested, I remove all the bonded devices, but this does not quite wipe all the state - for example, one observed difference is that immediately after flashing, before any device is paired, random address is regenerated on each boot.
As soon as a device is paired, the random address is stable and won't change, even if all the bonded devices are removed.

As a workaround, i'm currently using NVS functions to wipe `bt_config.conf/bt_cfg_key`. Maybe the solution is simply to add an official public API function to do that.

heyinling, 2017-12-21:
@rojer we can do remove bonded device by:

1. use `esp_ble_get_bond_device_list`to query bonded devices
2. then use `esp_ble_remove_bond_device` to remove the devices

These APIs are added after release v2.1. plz try on master or v3.0-rc1.

rojer, 2017-12-21:
i already do that. as i said, it doesn't seem to reset the device state completely (MAC address remains frozen).

heyinling, 2017-12-22:
> MAC address remains frozen

do you mean the local BT address didn't change? the local BT address can't be set by SW now and it's not stored in NVS.

rojer, 2017-12-22:
i use `esp_ble_gap_config_local_privacy(true)` to randomize the address.
as i said above, i observe the following behavior:
* freshly after flashing, before any pairing takes place, BT address is random and changes on every boot, which is to be expected
* as soon as device is paired, the address stops changing - also expected, however...
* when i remove all the bonded devices via the list/remove api as you suggested, the address remains frozen and does not change on subsequent reboots.

Yulong-espressif, 2017-12-25:
@rojer When the device is paired, If you exchange the IRK with the peer device, then the peer device will use your IRK exchange to resolve your local device address resolution, so even if your address has changed, the remote device also displays the original address. When you are bonded with the peer device, you act as a slave, and the peer device act as a master. So you should remove the bonded devices in the master device instead of slave device. Hope this answer can help you. Thanks.

closed 2018-03-15 by FayeY