espressif/esp-idf #1407: BLE security

chegewara, 2017-12-18:
What those means and why 8 < 8? Its connection with encryption between 2 esp32 devices. With my old samsung SGS4 its 6 < 8, im guessing its comparison BLE versions, but why 8 < 8?

`E (5077) BT: lmp_version_below LMP version 8 < 8`

Yulong-espressif, 2017-12-19:
The following definition are the LMP version table, help these can help you.

Yulong-espressif, 2017-12-19:
#define HCI_PROTO_VERSION     0x01      /* Version for BT spec 1.1          */
#define HCI_PROTO_VERSION_1_2 0x02      /* Version for BT spec 1.2          */
#define HCI_PROTO_VERSION_2_0 0x03      /* Version for BT spec 2.0          */
#define HCI_PROTO_VERSION_2_1 0x04      /* Version for BT spec 2.1 [Lisbon] */
#define HCI_PROTO_VERSION_3_0 0x05      /* Version for BT spec 3.0          */
#define HCI_PROTO_VERSION_4_0 0x06      /* Version for BT spec 4.0          */
#define HCI_PROTO_VERSION_4_1 0x07      /* Version for BT spec 4.1          */
#define HCI_PROTO_VERSION_4_2 0x08      /* Version for BT spec 4.2          */
#define HCI_PROTO_REVISION    0x000C    /* Current implementation version   */

negativekelvin, 2017-12-20:
https://github.com/espressif/esp-idf/blob/79f206be47c3f608615c1de8c491107e6c9194bb/components/bt/bluedroid/stack/smp/smp_act.c#L61

Warning is always printed before comparing version

chegewara, 2017-12-20:
Maybe it should be warning message logged, but as you can see its ERROR message logged.

negativekelvin, 2017-12-20:
https://github.com/espressif/esp-idf/blob/f8bda324ecde58214aaa00ab5e0da5eea9942aaf/components/bt/bluedroid/include/bt_trace.h#L30

All bt logs are routed to LOGE

But the comparison should be made before printing the warning

TianHao-Espressif, 2017-12-21:
@negativekelvin , due to that bluedroid has its own log system, but we also need to use system log, so we do BT_PRINTF as ESP_LOGE simply. I known it's need to be fixed.

chegewara, 2017-12-23:
Two esp32 devices, both should have `lmp_version_below LMP version 8 < 8` but one reports peer with level 6:

[image]

Yulong-espressif, 2017-12-25:
@chegewara Thanks for your log, I think this is really a  bug. We will fix it soon. Thanks again.

Yulong-espressif, 2018-03-14:
@chegewara We have check the log is not issue, 8 < 8 just means both the master & slave version are 4.2. Thanks.

chegewara, 2018-03-14:
@Yulong-espressif What about bluetooth v5.0 devices? With iPhone 8 log is reporting bluetooth v4.0:
`BT: lmp_version_below LMP version 6 < 8`

https://github.com/nkolban/esp32-snippets/issues/430#issuecomment-371997700

Yulong-espressif, 2018-03-27:
@chegewara What is the IOS system in your iphone 8?

chegewara, 2018-03-27:
@Yulong-espressif 11.2.6
https://github.com/nkolban/esp32-snippets/issues/430#issuecomment-376503697

chegewara, 2018-04-30:
@Yulong-espressif @Weijian-Espressif Any news here?
This is samsung S9+ with ble 5.0:
```
E (177069) BT: lmp_version_below LMP version 6 < 8
E (177559) BT: FOR LE SC LTK IS USED INSTEAD OF STK
```

Weijian-Espressif, 2018-05-02:
@chegewara this is a bug about version exchange, i have fixed it. The fix code will merge into IDF next week.

closed 2018-05-03 by chegewara

eminarcissus, 2018-06-12:
Can you reference a pull request to this issue? I would like to merge this into a old idf release

negativekelvin, 2018-06-12:
https://github.com/espressif/esp-idf/commits/master/components/bt/bluedroid/stack/smp/smp_act.c

eminarcissus, 2018-06-12:
thx very much, working on it :)