espressif/esp-idf #1408: gatts_exec_write_evt_param needs handle param

bpietsch, 2017-12-19:
There is currently no way to associate a ESP_GATTS_EXEC_WRITE_EVT event with the ESP_GATTS_WRITE_EVT events it follows. While I can presumably infer it from the order the events are received, this does not work well with the object oriented wrapper design I am working on.  Can you please add a handle parameter to gatts_exec_write_evt_param to make this possible?

heyinling, 2017-12-21:
@bpietsch according to my understanding the spec only allows one prepare write transaction at the same time on one connection. 

currently connection id is provided in ESP_GATTS_EXEC_WRITE_EVT  event. It provides enough info to find the prepare write transaction.

Weijian-Espressif, 2018-01-15:
@bpietsch ble only allows one prepare write at the same time on one connection.

closed 2018-01-15 by FayeY