espressif/esp-idf #1409: BLE security vulnerability

chegewara, 2017-12-19:
I am "working" and by working i mean i am trying to understand security mechanism in BLE. Ivegot some issues with my own application thats why i decided to perform tests on esp-idf provided examples. 
My authentication settings are for both, server and client:
- ESP_LE_AUTH_REQ_SC_ONLY
- ESP_IO_CAP_IO

Notised issue:
- 10/10 numeric comparison values are different but the same on both esp32 - just one answer is correct (YES)

My understanding is that on every reconnect numeric comparison should be the same or different, but 10 on 10 is not right.

Yulong-espressif, 2017-12-19:
@chegewara What is your problem? Can you description clearer, I still do not understand.

chegewara, 2017-12-19:
@Yulong-espressif 
If we are trying to connect 2 devices with ESP_IO_CAP_IO capability then each device should display numeric value to comparison with the other device. To perform authentication we need to confirm on both devices that 2 numbers are equal or not with buttons YES/NO. 

Im running examples from esp-idf (gatt_security_server and gatt_security_client) and every single time pseudo-generated numbers are the same on both devices. Which means to confirm authentication i have to press YES on both devices.

There should be about 50% chances that answer is either YES or NO. But we have 100% cases with answer YES.

chegewara, 2017-12-19:
Its from logs:
```
E (2837182) BT: Non bonding: No keys will be exchanged
E (2837182) BT: lmp_version_below LMP version 6 < 8
E (2837482) BT: Value for numeric comparison = 763988
```
763988 is value which is different every time im trying to connect 2 devices (2 x esp32) but its the same value on both esp32.

chegewara, 2017-12-19:
Im closing this issue because it seems my understanding was wrong. I have read it one more time and now i undertand that connection is performerd only when end user confirm with yes on both devices that sees the same numbers. My bad, sorry.

closed 2017-12-19 by chegewara