espressif/esp-idf #1410: make flash monitor issue

JPMJPM, 2017-12-19:
Hi,

Everything was ok but after the last Windows 10 actualitation "make flash monitor"
fails with this error -->

$ make flash monitor
Flashing binaries to serial port COM3 (app at offset 0x10000)...
esptool.py v2.1
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 18160 bytes to 10740...
Wrote 18160 bytes (10740 compressed) at 0x00001000 in 1.0 seconds (effective 152.3 kbit/s)...
Hash of data verified.
Compressed 468064 bytes to 298713...
Wrote 468064 bytes (298713 compressed) at 0x00010000 in 26.4 seconds (effective 141.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 82...
Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 1445.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
MONITOR
--- idf_monitor on COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock di
Traceback (most recent call last):
  File "C:/msys32/home/mainuser/esp/esp-idf/tools/idf_monitor.py", line 568, in <module>
    main()
  File "C:/msys32/home/mainuser/esp/esp-idf/tools/idf_monitor.py", line 502, in main
    monitor.main_loop()
  File "C:/msys32/home/mainuser/esp/esp-idf/tools/idf_monitor.py", line 269, in main_loop
    self.handle_serial_input(data)
  File "C:/msys32/home/mainuser/esp/esp-idf/tools/idf_monitor.py", line 300, in handle_serial_input
    self.console.write_bytes(b)
  File "C:\Python27\lib\site-packages\serial\tools\miniterm.py", line 63, in write_bytes
    self.byte_output.write(byte_string)
  File "C:/msys32/home/mainuser/esp/esp-idf/tools/idf_monitor.py", line 560, in write
    self.output.write(b)
IOError: [Errno 0] Error
make: *** [/home/mainuser/esp/esp-idf/components/esptool_py/Makefile.projbuild:96: monitor] Error 1

Any help ?, thanks.

negativekelvin, 2017-12-20:
https://github.com/espressif/esp-idf/issues/1136

JPMJPM, 2017-12-20:
Thanks negativekelvin, now it works fine.

closed 2017-12-20 by JPMJPM