espressif/esp-idf #1411: Sink connect issue

Elektrik1, 2017-12-20:
Hello,
I am developing an ESP32 BT audio player, and found such issue using provided A2DP example: if I connect from device, everything works fine, A2DP and AVRC are established fine; during successfull connection I save bt_addr to eeprom, and on next ESP boot I try to sink-connect to that device by calling esp_a2d_sink_connect(<saved_bt_addr>); It definitely does, but after few seconds disconnect occurs, with such log:
E (1060) BT_APP_CORE: bt_app_work_dispatch event 0x0, param len 0
E (2665) BT: reset flags
E (2675) BT_APP_CORE: bt_app_work_dispatch event 0x0, param len 16
E (2675) BT_AV: state 1, [08:21:ef:a4:97:eb]
E (3135) BT: btm_sec_connected

E (3195) BT: process_service_search_attr_rsp

E (3295) BT: bta_av_rc_create ACP handle exist for shdl:0
E (3335) BT_APP_CORE: bt_app_work_dispatch event 0x2, param len 16
E (3335) BT_AV: a2dp audio_cfg_cb , codec type 0
E (3335) BT_AV: configure audio player 21-15-2-35

E (3335) BT_AV: audio player configured, samplerate=44100
E (3365) BT: bta_dm_rm_cback:0, status:0
E (3375) BT_APP_CORE: bt_app_work_dispatch event 0x0, param len 16
E (3375) BT_AV: state 2, [08:21:ef:a4:97:eb]
E (3405) BT: process_service_search_attr_rsp

E (3435) BT: bta_av_rc_opened allow incoming AVRCP connections:1
E (3435) BT: bta_av_rc_opened rcb[1] shdl:1 lidx:1/0
E (3435) BT_AV: bt_app_rc_ct_cb eventass 0
E (3445) BT_APP_CORE: bt_app_work_dispatch event 0x0, param len 16
E (3445) BT_AV: avrc conn_state evt: state 1, feature 0x259, [08:21:ef:a4:97:eb]
E (3545) BT_APP_CORE: bt_app_work_dispatch event 0x2, param len 16
E (4975) BT: bta_av_proc_stream_evt: p_data is null
E (4975) BT: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x
824b

E (4975) BT: bta_dm_rm_cback:0, status:0
E (4985) BT: bta_dm_rm_cback:0, status:7
E (4985) BT: BTA_AV_START_EVT status 0, suspending 0, init 0

E (4995) BT_APP_CORE: bt_app_work_dispatch event 0x1, param len 16
E (5005) BT_AV: a2dp audio_state_cb state 2
E (36245) BT: bta_av_str_stopped:audio_open_cnt=1, p_data 0x0
E (36245) BT: bta_dm_rm_cback:0, status:6
**E (36245) BT: L2CAP - no LCB for L2CA_SetAclPriority**
**E (36245) BT: WARNING L2CA_SettFlushTimeout No lcb for bd_addr [...;a497eb]**
E (36255) BT: bta_dm_rm_cback:0, status:1
**E (36255) BT: ccb 1 not allocated**
E (36265) BT_APP_CORE: bt_app_work_dispatch event 0x1, param len 16
E (36265) BT_AV: a2dp audio_state_cb state 1
E (336275) BT_APP_CORE: bt_app_work_dispatch event 0x0, param len 16
E (36285) BT_AV: state 0, [08:21:ef:a4:97:eb]
E (36285) BT_AV: bt_app_rc_ct_cb eventass 0
E (36295) BT_APP_CORE: bt_app_work_dispatch event 0x0, param len 16
E (36295) BT_AV: avrc conn_state evt: state 0, feature 0x0, [00:00:00:00:00:00]

08:21:ef:a4:97:eb is that device bt_addr. It is clearly seen that bolded lines are causing device to be disconnected. What other functions should be called/configured before esp_a2d_sink_connect() is called ?

P.s. it worked fine on ESP-IDF 2.0, just updated it to 3.0, and this happened.

newkit, 2017-12-23:
Hi, trying to reproduce your issue here - would you mind sharing some code?

Elektrik1, 2017-12-26:
Sure, it's mostly a2dp_sink example, with such modifications:

    /* during init */
    uint8_t mac_to_c[6];

    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    if (esp_bt_controller_init(&bt_cfg) != ESP_OK) {
        ESP_LOGE(BT_AV_TAG, "%s initialize controller failed\n", __func__);
        return;
    }

    if (esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT) != ESP_OK) {
        ESP_LOGE(BT_AV_TAG, "%s enable controller failed\n", __func__);
        return;
    }

    if (esp_bluedroid_init() != ESP_OK) {
        ESP_LOGE(BT_AV_TAG, "%s initialize bluedroid failed\n", __func__);
        return;
    }

    if (esp_bluedroid_enable() != ESP_OK) {
        ESP_LOGE(BT_AV_TAG, "%s enable bluedroid failed\n", __func__);
        return;
    }

    /* create application task */
    bt_app_task_start_up();

    
    //**reads from NVS (reads fine, if print bda_t variable to Serial monitor)**
    nvs_handle my_handle1;
    nvs_open(STORAGE_NAMESPACE, NVS_READONLY, &my_handle1);
    size_t len = 0;
    nvs_get_str(my_handle1, "last_conn", NULL, &len);
    char* bda_t = (char*) malloc(len);
    nvs_get_str(my_handle1, "last_conn", bda_t, &len);
    nvs_close(my_handle1);

    for (int i = 0; bda_t[i] != 0; i++) {
        char byte[3];
        strncpy(byte, &bda_t[i * 2], 2); // copy 2 characters
        byte[2] = 0; // put NUL string terminator
        mac_to_c[i] = strtol(byte, NULL, 16); // convert string such as "e0" to 0xe0
    }
    
	
    
    /* Bluetooth device name, connection mode and profile set up */
    bt_app_work_dispatch(bt_av_hdl_stack_evt, BT_APP_EVT_STACK_UP, NULL, 0, NULL);
    //**sink connect call by mac, work, but disconnects after ~5sec.**
    esp_a2d_sink_connect(mac_to_c);

then in bt_app_av.c:

    static void bt_av_hdl_a2d_evt(uint16_t event, void *p_param) {
    ESP_LOGD(BT_AV_TAG, "%s evt %d", __func__, event);
    esp_a2d_cb_param_t *a2d = NULL;
    switch (event) {
        case ESP_A2D_CONNECTION_STATE_EVT:
        {
            a2d = (esp_a2d_cb_param_t *) (p_param);
            
            uint8_t *bda = a2d->conn_stat.remote_bda;
            ESP_LOGE(BT_AV_TAG, "state %d, [%02x:%02x:%02x:%02x:%02x:%02x]",
                    a2d->conn_stat.state, bda[0], bda[1], bda[2], bda[3], bda[4], bda[5]);

            
            //ESP_LOGE(BT_AV_TAG, "a2dp conn_state_cb, state %d, bd_addr:%08x%04x, in d: %d", a2d->conn_stat.state, (bda_tmp[0] << 24) + (bda_tmp[1] << 16) + (bda_tmp[2] << 8) + bda_tmp[3], (bda_tmp[4] << 8) + bda_tmp[5], bda_tmp);
            if (a2d->conn_stat.state == 0) {
                i2s_zero_dma_buffer((i2s_port_t) 0);
                i2s_stop((i2s_port_t) 0);

            } else if (a2d->conn_stat.state == 2) {
                //**saves to NVS on success connection (works fine)**
                nvs_handle my_handle;
                nvs_open(STORAGE_NAMESPACE, NVS_READWRITE, &my_handle);

                char *hexstrings = malloc(13);
                sprintf(hexstrings, "%08x%04x", (bda[0] << 24) + (bda[1] << 16) + (bda[2] << 8) + bda[3], (bda[4] << 8) + bda[5]);

                nvs_set_str(my_handle, "last_conn", hexstrings);
                nvs_commit(my_handle);
                nvs_close(my_handle);

            }
            break;
        }
        case ESP_A2D_AUDIO_STATE_EVT:
        {
            a2d = (esp_a2d_cb_param_t *) (p_param);
            ESP_LOGE(BT_AV_TAG, "a2dp audio_state_cb state %d", a2d->audio_stat.state);
            m_audio_state = a2d->audio_stat.state;
            if (ESP_A2D_AUDIO_STATE_STARTED == a2d->audio_stat.state) {
                m_pkt_cnt = 0;
            }
            break;
        }
        case ESP_A2D_AUDIO_CFG_EVT:
        {
            a2d = (esp_a2d_cb_param_t *) (p_param);
            ESP_LOGE(BT_AV_TAG, "a2dp audio_cfg_cb , codec type %d", a2d->audio_cfg.mcc.type);
            // for now only SBC stream is supported
            if (a2d->audio_cfg.mcc.type == ESP_A2D_MCT_SBC) {
                int sample_rate = 16000;
                char oct0 = a2d->audio_cfg.mcc.cie.sbc[0];
                if (oct0 & (0x01 << 6)) {
                    sample_rate = 32000;
                } else if (oct0 & (0x01 << 5)) {
                    sample_rate = 44100;
                } else if (oct0 & (0x01 << 4)) {
                    sample_rate = 48000;
                }
                i2s_set_clk(0, sample_rate, 16, 2);
                // MediaControl *ctrl = (MediaControl *)local_service->Based.instance;
                // CodecEvent evt;

                // ctrl->decoderRequestSetup(ctrl, (void *)&evt);
                // temporarily hardcoded the PCM configuaration
                ESP_LOGE(BT_AV_TAG, "configure audio player %x-%x-%x-%x\n",
                        a2d->audio_cfg.mcc.cie.sbc[0],
                        a2d->audio_cfg.mcc.cie.sbc[1],
                        a2d->audio_cfg.mcc.cie.sbc[2],
                        a2d->audio_cfg.mcc.cie.sbc[3]);
                ESP_LOGE(BT_AV_TAG, "audio player configured, samplerate=%d", sample_rate);
            }
            break;
        }
        default:
            ESP_LOGE(BT_AV_TAG, "%s unhandled evt %d", __func__, event);
            break;
    }
    }

Elektrik1, 2017-12-26:
Not sure what helped about it, but I have just pulled latest esp-idf from git, and it started to work fine (do not disconnect!
Can't tell which commit (in last few days) fixed it, but it now works.

One more strange behaviour: if in connected to it (or autoconnected) - everything works fine. If I do reset (pressing RST button, no power disruption), it hangs on log for like 15 seconds:
```
	E (767) BT_APP_CORE: bt_app_work_dispatch event 0x0, param len 0
	E (777) BT: reset flags
	E (777) BT_APP_CORE: bt_app_work_dispatch event 0x0, param len 16
	E (777) BT_AV: state 1, [08:21:ef:a4:97:eb]
	E (1187) BT: btm_sec_connected
```

and then connects:
```
	E (1277) BT: process_service_search_attr_rsp
	E (1377) BT: bta_av_rc_create ACP handle exist for shdl:0
	......
```

but if I restart once more, it connects immediately without hang. So basically fast connects are only happening every second restart. Same applies on initial cold boot - it hangs and connects, but if I manually restart during hang, it connects flash fast. Not sure if esp_a2d_sink_connect(mac_to_c); should be called somewhere in other place (like in some callback), and what causing it (maybe every second restart some services restart causes long delay some why).

Thanks for help, and whole crew working on this project.

newkit, 2017-12-29:
Thank you for your example. I couldn't get it to do a client initiated "cold" connect, where the client device connects to a previously "paired" but not currently connected ESP. I use the term paired in quotation marks since there is no actual pairing.
It also seems to depend on the client - when using an iPhone there was no reconnect, event when connected and doing a manual reset of the ESP. But when using a Mac, it did reconnect when manually resetting the ESP. What client do you use for your tests?

I also have seen the errornous behaviour you describe above where we see status 3.
So far I had have seen the client entering status 1 on the ESP, but not 2 as it would for a normal connection.

Elektrik1, 2017-12-29:
Thank you for your response, for tests I do use Android 7.0.1 (Galaxy S7). Will try with iPhone tomorrow and will feedback.

blueMoodBHD, 2018-03-22:
@Elektrik1 Is there still any problem now?

closed 2018-04-04 by FayeY

newkit, 2018-04-09:
@Elektrik1 Hey Julius, could you get your example to work autoconnecting?