espressif/esp-idf #1413: Feature Request: RFCOMM/SPP for Bluetooth Classic

LosDeiblos, 2017-12-20:
Will there be support for RFCOMM/SPP for classic Bluetooth in the near future?

blueMoodBHD, 2017-12-27:
Yes, we will support SPP for classic BT,and it is almost completed, will be released soon.

spmp, 2018-01-01:
@blueMoodBHD Can you give an indication of the timescale of soon and a link to the current branch/docs/headers/examples(if any) where this work is being done so we can get a head start please?
Cheers

blueMoodBHD, 2018-01-02:
@spmp We will merge to master in one or two week, please wait patiently, thanks for your understanding.

blueMoodBHD, 2018-01-09:
@spmp @LosDeiblos The SPP have merged into master, Please refer to esp-idf/examples/bluetooth/bt_spp_acceptor & esp-idf/examples/bluetooth/bt_spp_initiator

closed 2018-01-11 by FayeY