espressif/esp-idf #1415: Error compiling esp-hal-ledc.c in arduino library using Eclipse

jcherry4, 2017-12-20:
I am using Eclipse with the Arduino library and I get the following error when I make all:

_/home/jcherry/esp32/NetIrrigationContr/components/arduino/cores/esp32/esp32-hal-ledc.c:84:34: error: 'volatile union <anonymous>' has no member named 'div_num'
     LEDC_TIMER(group, timer).conf.div_num = div_num;//18 bit (10.8) This register is used to configure parameter for divider in timer the least significant eight bits represent the decimal part.
                                  ^_

I get the same error using ESP32_gpio_WS2812_Makuna and WiFi Web Server LED Blink  in the Adunio examples.
My Arduino template was downloaded from https://github.com/pcbreflux/espressif/tree/master/esp32/app/ESP32_Arduino_template

I am using a library from 
https://github.com/espressif/arduino-esp32

I am using  Oxygen.1a Release (4.7.1a) version of Eclipse
Running on Ubuntu in Vurital Box

Any help with this would be much appreciated.
Thanks
John

igrr, 2018-01-06:
`div_num` member has been removed some time ago, but recently added back (as deprecated):
https://github.com/espressif/esp-idf/blob/a1b59679f9d98a5a56bfd16fbab9c62b63518228/components/soc/esp32/include/soc/ledc_struct.h#L82

This field is also present in the latest version of arduino-esp32:
https://github.com/espressif/arduino-esp32/blob/70d0d4648766cea047613062ed50a9ecfc9de31f/tools/sdk/include/soc/soc/ledc_struct.h#L73

Please check that both arduino and esp-idf versions are recent enough.

jcherry4, 2018-01-07:
Hi Ivan,

Thanks for the info. I replaced ledc_struct.h in the esp-idf/components/soc/esp32/include/soc dir and that solved the problem.
I now get the following warning through the serial terminal, when it's running:

Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: loopTask

When I replaced the full esp-idf I got got a host of errors that indicated damage may occur to the ESP-32
I am using the  SimpleWIFIServer from the examples in in the Arduino examples.

Thanks

John

chegewara, 2018-01-07:
Im suspecting you have deleted delay() from main loop()

igrr, 2018-01-08:
@jcherry4 This message indicates that `loopTask` (which is the task where Arduino `setup` and `loop` functions run) takes all the CPU time on core 1, preventing IDLE task from running. Please have a look at task watchdog documentation: http://esp-idf.readthedocs.io/en/latest/api-reference/system/wdts.html#task-watchdog-timer

If you don't need to have the task watchdog checking IDLE task on CPU1, you can disable this in menuconfig:
http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#config-task-wdt-check-idle-task-cpu1

jcherry4, 2018-01-08:
Thanks Ivan,
Disabling the IDLE task solved it.
As I get more familiar with this device I will play around with the watchdog time out task.

Regards ,
John

closed 2018-02-12 by igrr