espressif/esp-idf #1416: AWS IoT MQTT constantly disconnects

seopyoon, 2017-12-21:
I have my own application, which keeps on disconnecting.  (It used to operate much longer without disconnection before -- not aware since which update)  

So, I tested with the AWS IoT subscribe_publish example from the esp-idf git repo.  And it still behaves similarly.

But when it disconnects, there is no way of knowing what caused it.  Could it be the WiFi / Router problem? Or is it the firmware? 

And also, at launch, it the handshake failes once or twice before successfully connecting.  (Does not take too long to establish connection though).

Please help!  Or, if anyone is experiencing a similar problem, please comment.

[image]

satishIoT, 2018-03-27:
@seopyoon 
pl check https://github.com/espressif/esp-idf/issues/1695
this suggestion form negativekelvin had worked for me.

projectgus, 2019-03-01:
Closing due to lack of update. Hopefully the linked issue solved the problem.

closed 2019-03-01 by projectgus