espressif/esp-idf #1418: Release/v3.0 ESP as GATT server, close GATT connection by server report incorrect conn_id.

Pines, 2017-12-21:
Sorry for bad grammar, I'm using ESP32 as GATT server.

And implement a keep-alive feature in ESP32, while a client phone inactive for a while, 
I would like manually disconnect this client, then ready accept other client connect in.

In such case, when keep-alive timer is timeout, I call esp_ble_gatts_close, it will report **conn_id** error finally in bta_gatts_close this line APPL_TRACE_ERROR("Unknown connection ID: %d", p_msg->hdr.layer_specific);

I trace the caller and callee stack, right in esp_ble_gatts_close, in arg.close.conn_id = BTC_GATT_CREATE_CONN_ID(gatts_if, conn_id); 
will resulting conn_id transfered to bta_gatts_close been combine of gatts_if and conn_id. 

Currently, I add this line 	p_msg->hdr.layer_specific&=0xFF; at start of bta_gatts_close to avoiding.

heyinling, 2017-12-22:
@Pines did you save the conn_id form ESP_GATTS_CONNECT_EVT and use it to close the connection?

Pines, 2017-12-22:
@heyinling 
Right， that's what I do.

For instance, in ESP_GATTS_CONNECT_EVT,
I have code like this ble_server_if = param->connect.server_if; ble_conn_id   = param->connect.conn_id;
then ble_gatts_if = 3, ble_conn_id =3.
that would be parameters I passing through to esp_ble_gatts_close( 3, 3).
the parameters transfer to bta_gatts_close, conn_id become (uint16_t)0x0303. combination of gatts_if and conn_id.

Pines, 2017-12-22:
Btw, I'm using Blufi now, current only one connection is permit at a moment.

heyinling, 2017-12-25:
@Pines I'm able to use `esp_ble_gatts_close ` to close the connection in my own application. Could you share your code and reproduce steps for this?

 BTW, The connection ID in protocol stack is different from application. maybe @Yulong-espressif can describe more about this.

Pines, 2017-12-25:
thanks you for replying @heyinling, I'm using code from BluFi, not sure could that be reason my result not same. 

here is related code, where conn_id been saved, where using it to close connection. any more hint you need, just let me know.

anyway, one more time thanks for your time.
/* ------------------------------------------------------------ */
// define 
#include "esp_gatts_api.h"
static uint8_t    ble_server_if;					/*!< server interface */
static uint16_t   ble_conn_id; 						/*!< Connection id */

/* ------------------------------------------------------------ */
/* on connected event */
    case ESP_BLUFI_EVENT_BLE_CONNECT:
        BLUFI_INFO("BLUFI ble connect");
        esp_ble_gap_stop_advertising();
        blufi_security_deinit();
        blufi_security_init();

	ble_server_if = param->connect.server_if;
	ble_conn_id   = param->connect.conn_id;

/* ------------------------------------------------------------ */
// timer
#include "freertos/timers.h"
#define BT_BLE_TIMEOUT				(2*60*1000/portTICK_RATE_MS)

static void ble_timer_callback( TimerHandle_t xTimer )
{
	if ( 0!=ble_server_if && 0!=ble_conn_id )
	{	
		ESP_LOGD( TAG, "%s BLE keep-alive is timeout!", __func__ );
		esp_ble_gatts_close( ble_server_if, ble_conn_id );
		
		ble_server_if = 0;
		ble_conn_id   = 0;
		/*  there is a bug timer will execute one shot after stop */
		if ( NULL!=ble_timer ) xTimerStop(ble_timer,0);
	}
}

int ble_timer_create()
{
	ble_timer = xTimerCreate( "bt_ble_timer", (TickType_t)BT_BLE_TIMEOUT, pdTRUE, (void *)0, ble_timer_callback );
	if( NULL==ble_timer ) 
	{ ESP_LOGE(TAG, "ble_timer timer create error!"); return 1; }
	
	return 0;
}

Yulong-espressif, 2017-12-25:
@Pines if ( 0!=ble_server_if && 0!=ble_conn_id ) ???
What is this code meaning? Do you want to call the esp_ble_gatts_close function when both ble_server_if and ble_conn_id are not illegal? However, ble_conn_id = 0 generally represents the first connection, so judging it in this way causes the first connection to never call the esp_ble_gatts_close function, which will result in the first connection not being able to disconnect. You should set that the value of ble_conn_id initialization is set to 0xffff instead of 0.

Pines, 2017-12-25:
@Yulong-espressif ,

I just replace no-connection to 0xFFFF, No server to 0xFF. and reset cpu, start a new connection, server_if is '3' conn is 3, didn't see it start from 0。and when calling esp_ble_gatts_close got E (48184) BT: Unknown connection ID: 771

Pines, 2017-12-25:
IDF version: v3.0-dev-806-gde750e99-dirty

Yulong-espressif, 2017-12-25:
@Pines your ESP32 multi connect to more then one device at the same time, Yet?

Yulong-espressif, 2017-12-25:
@Pines The conn id always start from 0

Pines, 2017-12-25:
only one device at a moment,
when it connect, 
        esp_ble_gap_stop_advertising();
        blufi_security_deinit();
        blufi_security_init();

stop advertising I guess suppose means phone can't  connect in anymore.
 I'm using BluFi example, not Gatt server example

Yulong-espressif, 2017-12-25:
@Pines So can you print the con_id when you receive the ESP_BLUFI_EVENT_BLE_CONNECT.
Do it like this:"BLUFI_ERROR("conn_id = %d\n", param->connect.conn_id);"

Yulong-espressif, 2017-12-25:
@Pines Can you also show the ble_connected( param ); function code to me?

Pines, 2017-12-25:
blufi_env.frag_size = BLUFI_FRAG_DATA_DEFAULT_LEN;		
	
        if ( pdPASS != xTimerStart( ble_timer, 500/portTICK_RATE_MS ) )
		ESP_LOGE(TAG, "%s ble timer start error", __func__ );

	ble_server_if = param->connect.server_if;
	ble_conn_id   = param->connect.conn_id;

Yulong-espressif, 2017-12-25:
I think you should use the esp_blufi_close to disconnect the blufi connection instead of use esp_ble_gatts_close. Because the meaning of conn_id in the blufi is different with other gatt services.

Yulong-espressif, 2017-12-25:
@Pines

Pines, 2017-12-25:
@Yulong-espressif  what my purpose is not change code in components, as such I will facing sync problem when I maintenance a difference version with ESP-IDF master.

If esp_blufi_close solve it, then this issue can be close.  conn_id weather start from 0 as GATT server, or 3 as BluFi is all OK to me.

Yulong-espressif, 2017-12-25:
@Pines We will also change the blufi conn_id start from 0 later. Thank you very much.

Pines, 2017-12-25:
@Yulong-espressif  Just swap to esp_blufi_close, it works fine. 
thanks for your insight knowledge.

By now I'm using other method setting up wifi SSID, password. 
BluFi module only working as GATT server, I also find some sequence number not expected.
So now I'm thinking of replace BluFi with a real GATT server, hoping it will improve sequence number not expected error. 

This topic is very well finished, I have got my expected answer,it can be close then.
anyway, do you have thought on change BluFi to GATT server? I would very glade to have your suggestion.

Pines, 2017-12-26:
@heyinling  @Yulong-espressif  thank you two for helping and patient.

closed 2017-12-26 by Pines