espressif/esp-idf #1419: SD card example is not responding

Rajkumar181, 2017-12-21:
Hello,
i tried all possible think, what you gave in example ,and i am getting the error. 
"I (10268) example: Using SPI peripheral
E (10268) example: Failed to initialize the card (259). Make sure SD card lines have pull-up resistors in place."
 so i change the  spi pins and some configuration,  than i tried once again  read and write the data into sd card. and i am success to read the data from SD card but the problem is , while writing the data into sd card. It is able to create a file but it is not writing on that.

My code :

sdmmc_host_t host = SDSPI_HOST_DEFAULT();
host.slot = VSPI_HOST;
host.flags = SDMMC_HOST_FLAG_SPI;
host.max_freq_khz = 400; 

sdspi_slot_config_t slot_config = SDSPI_SLOT_CONFIG_DEFAULT();
slot_config.gpio_miso = PIN_NUM_MISO;
slot_config.gpio_mosi = PIN_NUM_MOSI;
slot_config.gpio_sck = PIN_NUM_CLK;
slot_config.gpio_cs = PIN_NUM_CS;
slot_config.gpio_cd = SDMMC_SLOT_NO_CD;
slot_config.gpio_wp = SDMMC_SLOT_NO_WP; 

esp_vfs_fat_sdmmc_mount_config_t mount_config = {
.format_if_mount_failed = true,
.max_files = 5
};

sdmmc_card_t* card;
esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &card);

if (ret != ESP_OK) {
if (ret == ESP_FAIL) {
ESP_LOGE(TAG, "Failed to mount filesystem. "
"If you want the card to be formatted, set format_if_mount_failed = true.");
} else {
ESP_LOGE(TAG, "Failed to initialize the card (%d). "
"Make sure SD card lines have pull-up resistors in place.", ret);
}
return;
}

// Card has been initialized, print its properties
sdmmc_card_print_info(stdout, card);

// Use POSIX and C standard library functions to work with files.
// First create a file.
#define PATH1 "/sdcard/hello.txt"

ESP_LOGI(TAG, "Opening file");
f = fopen(PATH1, "w");
if (f == NULL) {
ESP_LOGE(TAG, "Failed to open file for writing");
return;
}else
ESP_LOGI(TAG, "Success to open file for writing");
ret = fprintf(f, "Hello %s!\n", card->cid.name); 
ESP_LOGI(TAG, "closing..");
fclose((FILE*)f);
ESP_LOGI(TAG, "closed");
esp_vfs_fat_sdmmc_unmount();
ESP_LOGI(TAG, "Card unmounted");

Error Result : 

Name: N/A 
Type: SDSC
Speed: default speed
Size: 1921MB
CSD: ver=0, sector_size=512, capacity=3934208 read_bl_len=10
SCR: sd_spec=2, bus_width=5

I (11188) example: Opening file
V (11188) vfs_fat: vfs_fat_open: path="/hello.txt", flags=601, mode=1b6
I (11198) example: Success to open file for writing
I (11208) example: closing..
V (11208) sdmmc_cmd: sending cmd slot=2 op=24 arg=3c9e00 flags=1c10 data=0x3ffb64b4 blklen=512 datalen=512 timeout=5000
V (11218) sdspi_host: sdspi_host_start_command: slot=2, CMD24, arg=0x003c9e00 flags=0x7, data=0x3ffb64b4, data_size=512 crc=0x0e
D (11238) sdspi_host: start_command_write_blocks: invalid R1 response: 0xff
E (11238) sdspi_host: sdspi_host_start_command: cmd=24 error=0x108
D (11248) sdmmc_cmd: sdmmc_req_run returned 0x108
E (11248) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x108
E (11258) diskio_sdmmc: sdmmc_write_blocks failed (264)
D (11268) vfs_fat: vfs_fat_write: fresult=1
V (11268) sdmmc_cmd: sending cmd slot=2 op=24 arg=3c9e00 flags=1c10 data=0x3ffb64b4 blklen=512 datalen=512 timeout=5000
V (11278) sdspi_host: sdspi_host_start_command: slot=2, CMD24, arg=0x003c9e00 flags=0x7, data=0x3ffb64b4, data_size=512 crc=0x0e
D (11298) sdspi_host: start_command_write_blocks: invalid R1 response: 0xff
E (11298) sdspi_host: sdspi_host_start_command: cmd=24 error=0x108
D (11308) sdmmc_cmd: sdmmc_req_run returned 0x108
E (11308) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x108
E (11318) diskio_sdmmc: sdmmc_write_blocks failed (264)
D (11318) vfs_fat: vfs_fat_close: fresult=1
I (11328) example: closed
I (11328) example: Card unmounted

igrr, 2018-01-06:
There was a recent fix for SDSPI error 0x108: https://github.com/espressif/esp-idf/commit/269486ca4dcb335898405439905558206e014635. Which version of ESP-IDF are you getting the issue with?

Alvin1Zhang, 2019-10-12:
@Rajkumar181 Thanks for reporting, and sorry for the very slow turnaround. Would you please try whether this issue has been fixed? Many thanks.

Alvin1Zhang, 2020-06-11:
@Rajkumar181 Thanks for reporting. Would you please try whether this issue has been fixed? Thanks.

tom-borcin, 2021-02-25:
Hello @Rajkumar181,

thank you for reporting this issue. At Espressif we really appreciate it when our users report issues as it helps us make our products better. Unfortunately there hasn't been any progress with this issue for the last 3 years and we are closing it to keep our backlog manageable.

If you don't agree with this decision, please feel free to reopen the issue or contact me directly.

Regards,

Tomas

closed 2021-02-25 by tom-borcin