espressif/esp-idf #1421: GPIO configured as interrupt issue (for IO12)

hvishu, 2017-12-22:
GPIO configured as interrupt
if (pin is GPIO_NUM_21(21)) below code works fine (hitting ISR)

        gpio_config_t io_conf;
        //disable interrupt
        io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
        //disable pull-down mode
        io_conf.pull_down_en = 0;
        //disable pull-up mode
        io_conf.pull_up_en = 0;
        //configure GPIO with the given settings
        //change gpio intrrupt type for one pin
        gpio_set_intr_type(pin, GPIO_INTR_POSEDGE);
        //bit mask of the pins, use GPIO4/5 here
        io_conf.pin_bit_mask = (1 << pin);
        //set as input mode    
        io_conf.mode = GPIO_MODE_INPUT;
        gpio_config(&io_conf);

        //install gpio isr service
        gpio_install_isr_service(0);
        //hook isr handler for specific gpio pin
        gpio_isr_handler_add(pin, funcPtr, (void*) pin);

But above code does not work for GPIO_NUM_12(12).
        gpio_config_t io_conf;
        //disable interrupt
        io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
        //disable pull-down mode
        io_conf.pull_down_en = 0;
        //disable pull-up mode
        io_conf.pull_up_en = 0;
        //configure GPIO with the given settings
        //change gpio intrrupt type for one pin
        gpio_set_intr_type(pin, GPIO_INTR_POSEDGE);
        //bit mask of the pins, use GPIO4/5 here
        io_conf.pin_bit_mask = (1 << pin);
        //set as input mode    
        io_conf.mode = GPIO_MODE_INPUT;
        gpio_config(&io_conf);

        //install gpio isr service
        gpio_install_isr_service(0);
        //hook isr handler for specific gpio pin
        gpio_isr_handler_add(pin, funcPtr, (void*) pin);

    To get it working need to change 
            io_conf.pin_bit_mask = (1 << pin);
        to
            io_conf.pin_bit_mask = 0x8;

for IO12, working case below is the log
        Welcome to minicom 2.7

        OPTIONS: I18n 
        Compiled on Feb  7 2016, 13:37:27.
        Port /dev/ttyUSB0, 16:43:04

        Press CTRL-A Z for help on special keys

        ets Jun  8 2016 00:22:57

        rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
        configsip: 0, SPIWP:0xee
        clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
        mode:DIO, clock div:2
        load:0x3fff0018,len:4
        load:0x3fff001c,len:5752
        load:0x40078000,len:0
        ho 12 tail 0 room 4
        load:0x40078000,len:13804
        entry 0x40079030
        I (73) boot: Detected ESP32
        I (34) boot: ESP-IDF v3.0-dev-1187-g09d2791 2nd stage bootloader
        I (34) boot: compile time 16:38:28
        I (34) boot: Enabling RNG early entropy source...
        I (40) boot: SPI Speed      : 40MHz
        I (44) boot: SPI Mode       : DIO
        I (48) boot: SPI Flash Size : 4MB
        I (52) boot: Partition Table:
        I (55) boot: ## Label            Usage          Type ST Offset   Length
        I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
        I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
        I (78) boot:  2 factory          factory app      00 00 00010000 00100000
        I (85) boot: End of partition table
        I (89) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0bc70 ( 48240) map
        I (115) esp_image: segment 1: paddr=0x0001bc98 vaddr=0x3ffb0000 size=0x0331c ( 13084) load
        I (120) esp_image: segment 2: paddr=0x0001efbc vaddr=0x40080000 size=0x00400 (  1024) load
        I (122) esp_image: segment 3: paddr=0x0001f3c4 vaddr=0x40080400 size=0x00c4c (  3148) load
        I (132) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x1c5d8 (116184) map
        I (180) esp_image: segment 5: paddr=0x0003c5f8 vaddr=0x4008104c size=0x08c48 ( 35912) load
        I (195) esp_image: segment 6: paddr=0x00045248 vaddr=0x400c0000 size=0x00000 (     0) load
        I (201) boot: Loaded app from partition at offset 0x10000
        I (201) boot: Disabling RNG early entropy source...
        I (204) cpu_start: Pro cpu up.
        I (208) cpu_start: Starting app cpu, entry point is 0x40081100
        I (0) cpu_start: App cpu up.
        I (219) heap_init: Initializing. RAM available for dynamic allocation:
        I (225) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
        I (231) heap_init: At 3FFB3F40 len 0002C0C0 (176 KiB): DRAM
        I (238) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
        I (244) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
        I (250) heap_init: At 40089C94 len 0001636C (88 KiB): IRAM
        I (257) cpu_start: Pro cpu start user code
        I (274) cpu_start: Starting scheduler on PRO CPU.
        I (0) cpu_start: Starting scheduler on APP CPU.
        Hello world!
        This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
        Restarting now.
        Timer ISR init is done
        I (4365) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
        Cust.Salt_Trip_Days = 88
        Detected power interrupt    --> print message from ISR

for IO12, non-working case below is the log
        Welcome to minicom 2.7

        OPTIONS: I18n 
        Compiled on Feb  7 2016, 13:37:27.
        Port /dev/ttyUSB0, 16:50:08

        Press CTRL-A Z for help on special keys

        ets Jun  8 2016 00:22:57

        rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
        configsip: 0, SPIWP:0xee
        clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
        mode:DIO, clock div:2
        load:0x3fff0018,len:4
        load:0x3fff001c,len:5752
        load:0x40078000,len:0
        ho 12 tail 0 room 4
        load:0x40078000,len:13804
        entry 0x40079030
        I (73) boot: Detected ESP32
        I (34) boot: ESP-IDF v3.0-dev-1187-g09d2791 2nd stage bootloader
        I (34) boot: compile time 16:38:28
        I (34) boot: Enabling RNG early entropy source...
        I (40) boot: SPI Speed      : 40MHz
        I (44) boot: SPI Mode       : DIO
        I (48) boot: SPI Flash Size : 4MB
        I (52) boot: Partition Table:
        I (55) boot: ## Label            Usage          Type ST Offset   Length
        I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
        I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
        I (78) boot:  2 factory          factory app      00 00 00010000 00100000
        I (85) boot: End of partition table
        I (89) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0bc54 ( 48212) map
        I (115) esp_image: segment 1: paddr=0x0001bc7c vaddr=0x3ffb0000 size=0x0331c ( 13084) load
        I (120) esp_image: segment 2: paddr=0x0001efa0 vaddr=0x40080000 size=0x00400 (  1024) load
        I (122) esp_image: segment 3: paddr=0x0001f3a8 vaddr=0x40080400 size=0x00c68 (  3176) load
        I (132) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x1c5c8 (116168) map
        I (180) esp_image: segment 5: paddr=0x0003c5e8 vaddr=0x40081068 size=0x08c2c ( 35884) load
        I (195) esp_image: segment 6: paddr=0x0004521c vaddr=0x400c0000 size=0x00000 (     0) load
        I (201) boot: Loaded app from partition at offset 0x10000
        I (201) boot: Disabling RNG early entropy source...
        I (204) cpu_start: Pro cpu up.
        I (208) cpu_start: Starting app cpu, entry point is 0x40081100
        I (0) cpu_start: App cpu up.
        I (219) heap_init: Initializing. RAM available for dynamic allocation:
        I (225) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
        I (231) heap_init: At 3FFB3F40 len 0002C0C0 (176 KiB): DRAM
        I (238) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
        I (244) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
        I (250) heap_init: At 40089C94 len 0001636C (88 KiB): IRAM
        I (257) cpu_start: Pro cpu start user code
        I (274) cpu_start: Starting scheduler on PRO CPU.
        I (0) cpu_start: Starting scheduler on APP CPU.
        Hello world!
        This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
        Restarting now.
        Timer ISR init is done
        I (4355) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
        Guru Meditpa``&�2 ��Ғ��Ҫ�j

        rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
        configsip: 0, SPIWP:0xee
        clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
        mode:DIO, clock div:2
        load:0x3fff0018,len:4
        load:0x3fff001c,len:5752
        load:0x40078000,len:0
        ho 12 tail 0 room 4
        load:0x40078000,len:13804
        entry 0x40079030
        I (74) boot: Detected ESP32
        �����`ph

But I see using bit 3 instead of bit 12 is not correct.  Could you check?

kewalmshah, 2017-12-26:
Hi @hvishu can I know which esp32 development board you are using ?

koobest, 2018-01-03:
Hi @hvishu, is this problem solved? if not, can you tell me what board are you using?

koobest, 2018-01-08:
Hi, @hvishu,
    I'm sorry that I have failed to reproduce the problem with the information you provide. GPIO12 works fine here. can you check if GPIO12 is used for other purposes in your board?

hvishu, 2018-01-09:
Dear @koobest,
I have "ESP32 Core Board V2 / ESP32 DevKitC" and will test it.  Could you send me the code which you used to test the reported problem?
Regards,
Vishwa

koobest, 2018-01-09:
Hi @hvishu ,
my test code is very simple.

`#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/gpio.h"

#define GPIO_PIN   (12)

static void IRAM_ATTR funcPtr(void* arg)
{
    uint32_t gpio_num = (uint32_t) arg;
    ets_printf("gpio%d triggered\n", gpio_num);
}

void app_main()
{
    gpio_num_t pin = GPIO_PIN;
    gpio_config_t io_conf;
    io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
    io_conf.pull_down_en = 0;
    io_conf.pull_up_en = 0;
    io_conf.pin_bit_mask = (0x1 << GPIO_PIN);
    io_conf.mode = GPIO_MODE_INPUT;
    gpio_config(&io_conf);
    gpio_set_intr_type(pin, GPIO_INTR_POSEDGE);
    gpio_install_isr_service(0);
    gpio_isr_handler_add(pin, funcPtr, (void*) pin);
    while(1) {
        vTaskDelay(1000 / portTICK_RATE_MS);
    }
}`

koobest, 2018-01-09:
@hvishu,  If the problem is still not resolved, please provide a test code that can reproduces the problem. thanks!!

koobest, 2018-01-10:
Hi, @hvishu
Did my test code can reproduce your problem?

hvishu, 2018-01-11:
Dear @koobest,
Was busy with some other stuff and not able to your code so far.  WIll test it within this week and let you know.
Regards,
Vishwa

koobest, 2018-01-23:
Hi, @hvishu, Is there any progress for this problem? 
leave a comment if you have any questions. 
thanks!!

FayeY, 2018-01-26:
Closing due to lack of response, please feel free to reopen if this is still an issue.

closed 2018-01-26 by FayeY

hvishu, 2018-02-06:
Sorry for the late reply.  
Guess, there was some mistake in configuring the interrupt from the side ( I am using two gpio's as interrupts), it's working fine.  Sorry for the trouble and thank you for responding.