espressif/esp-idf #1422: How to make multiple pins as input/output in one line?

sjoteppagol, 2017-12-22:
I am just trying to declare multiple pins as inputs or outputs.  I am not exactly getting dedicated API for that. In my application I am using plenty of IO's and every time declare as to gpio_pad_select_gpio(pin number) and so on, I feel is unconventional method.

`#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

#define INPUT_GPIO_NO 34  
unsigned int GPIOlevel;

void blink_task(void *pvParameter)
{
    gpio_pad_select_gpio(INPUT_GPIO_NO);
    gpio_pullup_dis(INPUT_GPIO_NO);
    gpio_set_direction(INPUT_GPIO_NO, GPIO_MODE_INPUT);
    
	while(1) 
	{
        GPIOlevel = gpio_get_level(INPUT_GPIO_NO);
		if (GPIOlevel == 1)
		{
			printf("GPIO Level is HIGH !!!\n");
		}
		else
        {			
			printf("GPIO Level is LOW \n");
        }
		vTaskDelay(100 / portTICK_PERIOD_MS);              
	}
}

void app_main()
{
    xTaskCreate(&blink_task, "blink_task", configMINIMAL_STACK_SIZE, NULL, 5, NULL);
}
`
Is there any API or function through which I can pass Pin number and mode at a time !?

negativekelvin, 2017-12-24:
gpio_config

kewalmshah, 2017-12-24:
You can use gpio_config_t and use bit mask. Checkout Gpio example code in esp-idf here: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/gpio/main/gpio_example_main.c

FayeY, 2017-12-26:
Hi @sjoteppagol, did the above suggestion help you resolve the issue?

sjoteppagol, 2017-12-27:
Here is simple example, lets assume we have 2 inputs BUTTON1 and BUTTON2 connected to Pin no 32 and 35. We have 2 output's LED1 and LED2 connected to pin 5 and 25 respectively. 
My question is instead of initializing these pins separately as shown in gpio_init(), is there any API through which I can pass pin and mode as a parameters and initialize all pins in one line. 
```
#define BUTTON1 GPIO_NUM_32
#define BUTTON2 GPIO_NUM_35
#define LED1 GPIO_NUM_5
#define LED2 GPIO_NUM_25
void gpio_init()
{
   gpio_pad_select_gpio(BUTTON1);
   gpio_set_direction(BUTTON1, GPIO_MODE_INPUT);
   gpio_pad_select_gpio(BUTTON2);
   gpio_set_direction(BUTTON2, GPIO_MODE_INPUT);

   gpio_pad_select_gpio(LED1);         
   gpio_set_direction(LED1, GPIO_MODE_OUTPUT);
   gpio_pad_select_gpio(LED2 );        
   gpio_set_direction(LED2 , GPIO_MODE_OUTPUT);
}
```

closed 2018-01-04 by sjoteppagol

kmmaran, 2019-07-04:
Hi sjoteppagol, I understand you close this issue now, but have you find a way to perform multiple gpio initialisation, or one after another.