espressif/esp-idf #1423: Calling esp_ota_begin with low memory break the factory partion

CPZbre, 2017-12-22:
**Version ESP-IDF 2.1.1**

We have an application on ESP-WROOM than use BLE and WIFI with TLS.
When we call the esp_ota_begin function we have only 55840 bytes in heap we have this traces :

> W (45932) Communication: Free heap size: 55840 (>>> esp_ota_begin)
> Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
> Register dump:
> PC      : 0x40189b61  PS      : 0x00060e30  A0      : 0x00000000  A1      : 0x3ffdb2b0
> 0x40189b61: ppTask at ??:?
> 
> A2      : 0x3ffcf7a0  A3      : 0x3ffc8448  A4      : 0x0000001a  A5      : 0x40189c27
> 0x40189c27: ppTask at ??:?
> 
> A6      : 0x3f425b5c  A7      : 0x00000000  A8      : 0x80189b5c  A9      : 0x3ffdb270
> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffd9d08  A13     : 0x00000000
> A14     : 0x3ffda25c  A15     : 0x3ffc9100  SAR     : 0x00000010  EXCCAUSE: 0x00000000
> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
> 
> Backtrace: 0x40189b61:0x3ffdb2b0 0x7ffffffd:0x3ffdb2f0
> 0x40189b61: ppTask at ??:?
> 
> 
> Rebooting...
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0x00
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0008,len:8
> load:0x3fff0010,len:3432
> load:0x40078000,len:10936
> load:0x40080000,len:796
> 0x40080000: _WindowOverflow4 at ??:?
> 
> entry 0x40080254
> 0x40080254: _Level5Vector at ??:?
> 
> **E (1671) esp_image: checksum failed. Calculated 0x57 read 0xfa**
> **E (1672) boot: Failed to verify app image @ 0x10000 (8194)**
> user code done

At this step the BLE api is not active juste initialized.

We have the same code and sequence in an other application but without BLE api include and in this case it is working well.

FayeY, 2017-12-26:
Hi, when this happened, could the ESP32 still work after power reset?

CPZbre, 2018-01-03:
It's leave in checksum failed

malaimoo, 2018-01-04:
You should upgrade your IDF to 3.0 rc.1, The ble and wifi allocate less ram , and the ota will work well under ble&wifi.
But, 3.0rc1 is also not very good to solve this completely, It will stuck sometimes in my tests.

projectgus, 2018-01-04:
Hi @CPZbre ,

Are you able to share some of your flash dumps and .bin/.elf files with us, please? You can use `esptool.py -p PORT read_flash 0 0x400000 flash_dump.bin` for a dump of the invalid flash. If you can send the .bin and .elf files from the OTA update build directory as well, this would be good. Also the original factory image if you have it. If you don't want to attach these publically then you can email them to angus at espressif.

Is the bootloader you are using from IDF v2.1.1 as well, or an older version?

Angus

projectgus, 2018-04-02:
Closing due to a lack of information. @CPZbre , feel free to reopen if you still have the information requested above.

closed 2018-04-02 by projectgus