espressif/esp-idf #1426: mbedtls hardware acceleration not enough memory

ifullgaz, 2017-12-23:
I'm using mbedtls to implement a the SRP 6 algorithm. I've hit an issue with the memory needed to calculate the premaster secret as per RFC5054 (S = (A * v^u) ^ b % N).
In my scenario, (A * v^u) has 192 bytes which would require 192 * 32 = 6144 bytes in the register but the limit is set at 4092 bytes.
Specifically, I get a MBEDTLS_ERR_MPI_NOT_ACCEPTABLE error from mbedtls_mpi_exp_mod.
Having tried to play with the esp_bignum.c and hwcrypto_reg.h files, I realised that the solution is not as simple as just playing with the numbers...
Can this limit be changed or is it strictly hardware dependant?

projectgus, 2018-01-08:
Hi @ifullgaz ,

Sorry for the delay in getting back to you.

The hardware limit is real. There's no way that the registers for modular exponentiation can hold more than 4096 bit arguments.

It is probably possible to decompose the modexp operation into multiple hardware operations. We follow a similar approach for bignum multiplication when either argument is longer than 2048 bits:
https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/esp_bignum.c#L629

However, probably the quickest and easiest solution is to disable hardware bignum acceleration.

Angus

Alvin1Zhang, 2021-02-02:
Thanks for reporting, and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/ea79091725e0e6c74b54bae794ffef78c5b4c089, feel free to reopen.

closed 2021-02-02 by Alvin1Zhang