espressif/esp-idf #1431: mDNS does not work in ethernet mode due to a harcoded check

dirkx, 2017-12-24:
One should perhaps add below. It is probably even better if this check is removed (so that mDNS works automatically for all network types). As that avoids having to pass down the interface all the way through OTA and similar distant user.

Dw.

diff --git a/components/mdns/mdns.c b/components/mdns/mdns.c
index 171da17..416e946 100644
--- a/components/mdns/mdns.c
+++ b/components/mdns/mdns.c
@@ -1602,17 +1602,18 @@ esp_err_t mdns_init(tcpip_adapter_if_t tcpip_if, mdns_server_t ** mdns_server)
         return ESP_ERR_INVALID_STATE;
     }
 
-    uint8_t mode;
-    err = esp_wifi_get_mode((wifi_mode_t*)&mode);
-    if (err) {
-        return err;
-    }
-
-    if ((tcpip_if == TCPIP_ADAPTER_IF_STA && !(mode & WIFI_MODE_STA))
-            || (tcpip_if == TCPIP_ADAPTER_IF_AP && !(mode & WIFI_MODE_AP))) {
-        return ESP_ERR_INVALID_ARG;
+    if (tcpip_if != TCPIP_ADAPTER_IF_ETH) {
+        uint8_t mode;
+        err = esp_wifi_get_mode((wifi_mode_t*)&mode);
+        if (err) {
+            return err;
+        }
+    
+        if ((tcpip_if == TCPIP_ADAPTER_IF_STA && !(mode & WIFI_MODE_STA))
+                || (tcpip_if == TCPIP_ADAPTER_IF_AP && !(mode & WIFI_MODE_AP))) {
+            return ESP_ERR_INVALID_ARG;
+        }
     }
-
     mdns_server_t * server = (mdns_server_t *)malloc(sizeof(mdns_server_t));
     if (!server) {
         return ESP_ERR_NO_MEM;

FayeY, 2018-01-03:
@me-no-dev , could you help with this issue? Many thanks.

me-no-dev, 2018-02-13:
Actually the interface needs to be passed and I did test with ETH. What is the issue? Did you pass the system events to mDNS? Each interface is handled separately, we can not use a global socket for all, because we need to know on which interface the packet came and on which interface to send it back. With multicast interfaces LwIP messes up and does not know where to route the packet.

dirkx, 2018-02-13:
Yes - I am (was) passing it (as early as possible; i.e. in MDNS.begin(host, iface); as OTA or something enables mDNS and in mdns_init (with the right wired ethernet passed, no wifi set up) it then immediately hits - esp_wifi_get_mode(). 

Which gets unhappy in the ETH only (with no wifi of WiFi.h included) case.

Dw.

me-no-dev, 2018-02-13:
Wait... :D I have the feeling you are talking about the old code? mDNS has drastically changed recently (though the Arduino API stays the same)

dirkx, 2018-02-13:
The two patches (#994 and #1432) where against HEAD/tip of master as of the day of the patch.

david-cermak, 2019-03-27:
Hi @dirkx 

Can we close this issue? It seems you were referring to the older code, which has been updated and this ethernet issue has been resolved differently.

I have just tested mdns with ethernet as a separate interface and together with wifi as duplicated interface and both work as expected.

dirkx, 2019-03-27:
Tested against the HEAD. Things seem to work for the WROOM32 (but not for the Olimex -- likely an issue on their end) -- except for the fact that the mDNS now stops after a few days of continuous running. Or rather - the announcement and 5353 UDP resolution stops to respond. 

Fair to assueme that this is a totally isolated regression ?

In that case fair to close this - and I'll raise a new timer related one on the mDNS dying after 37~60 hours (not pinpointed the preceice date - as I got also hit by #2455 in parallel.

david-cermak, 2019-03-27:
ok, thanks, could you please close it? 

mdns stopped working after several hours must be a separate issue, please raise a new one. I'll test it also on my end.

closed 2019-03-27 by dirkx