espressif/esp-idf #1432: JTAG stepping through code ... SIGTRAP in xQueueGenericReceive

nkolban, 2017-12-24:
While using JTAG with a WROVER DevKit V3, I am able to step through my code ... however at random times in any given session, my debugging stops with gdb jumping me to a SIGTRAP reported as:

```
Program received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 1073412716]
xQueueGenericReceive (xQueue=0x3ffaea14, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0)
    at /home/kolban/particle_esp32/esp-idf-v3.0-rc1/components/freertos/./queue.c:1455
1455			taskENTER_CRITICAL(&pxQueue->mux);
(gdb) 
```

For example ...
If I start my app using gdb and then place a break at main and continue the app, I wake up in main just fine.  If I then step over the next statements after some time (it may be the first statement, it may be the 10th statement), this SIGTRAP occurs.

Is there a recipe to ignore this or otherwise properly handle it?  I can recreate at will and will be delighted to provide any diagnostics needed.

igrr, 2017-12-25:
Hi @nkolban, this part of the documentation mentions how to generate log files from OpenOCD and GDB: http://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/tips-and-quirks.html#jtag-debugging-tip-reporting-issues

nkolban, 2017-12-25:
Thank you sir.  Will review and repost the extra details within 24 hours.  Many thanks again.

nkolban, 2017-12-25:
Closing issue here ... and will open on the correct repository found here:

https://github.com/espressif/openocd-esp32/issues

closed 2017-12-25 by nkolban