espressif/esp-idf #1433: Simple Wi-Fi example

xarrange, 2017-12-25:
Hi, I believe its worth to add simple wi-fi examples to getting-started.

By "simple" I mean:
1) STA:  connect to AP using provided SSID/pass and print ip addr
2) AP:  using provided SSID/pass create AP, wait till client connects, print ip addr
This should be like "hello world" in terms of wi-fi

It's not so comfortable to cut pieces of code from several examples - you'll always have background thought "I missed something"

copercini, 2017-12-29:
Take a look on Arduino for ESP32, it's a simplification layer that runs over IDF: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClient/WiFiClient.ino
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

xarrange, 2017-12-29:
That's even worse. To use this Arduino example in pure ESP-IDF program I have to change much more things.

TimXia, 2018-01-04:
@xarrange Thanks for your kindly suggestion. We will add this example.

closed 2018-03-19 by igrr