espressif/esp-idf #1434: BLE event ESP_GATTC_WRITE_CHAR_EVT returns status ESP_GATT_OK when truncated

nkolban, 2017-12-25:
Imagine we have a BLE client application that is writing new values to a remote BLE server.  It is using the API called `esp_ble_gattc_write_char`.  This results in an eventual callback indicating the response.  This is returned as a BLE GATTC event of type `ESP_GATTC_WRITE_CHAR_EVT`.

Now imagine I am sending a data payload of 40 bytes.  If the MTU defined by the server is the default of 23 bytes then the message is too big to be transmitted and we get an error indication in the console:

`E (10273) BT: attribute value too long, to be truncated to 20`

What I am finding is that the message is still written to the server (in its truncated form) and we get an `ESP_GATTC_WRITE_CHAR_EVT` as expected and its status property is `ESP_GATT_OK`.  To me, this means that the client application that thought it sent 40 bytes is un-aware that this was too big for the MTU and is not informed.   I would have expected a different status indicating sent truncated rather than a simple "OK".   This talks to a semantics question.  Is the current implementation the correct behavior?

Yulong-espressif, 2017-12-26:
@nkolban Thank you for your good question, In fact, when you send less than 600 bytes of data on the application, we will determine whether the current data has exceeded the MTU size on the gatt stack. If the MTU size is exceeded, we split the data into multiple data packets, Use multiple preparewrite and execute write to send, and print log to remind the user at this time the data is divided into several sending end-user, instead of sending. The purpose of this design is to allow users to use more simple, when sending large data packets, it does not require the user to manually data segmentation. We will aim to set the maximum packet 600 is given to the cause of memory. So what if the user wants to send more than 600 bytes of data (and the MTU is less than 600)? At this time the user can only be called in the application prepare write sent. Note that when prepare write is called to send data, the data size can not exceed the MTU. At this time, because the application on the user manual data segmentation, so the protocol stack will no longer automatically help users data segmentation.

chegewara, 2017-12-26:
@Yulong-espressif If this is true then we have bug here because im sending 25 bytes and nRF is receiving only 20 as expected since BT stack is truncating message to 20 bytes.

chegewara, 2017-12-26:
@Yulong-espressif after very thorough investigating its what ive found:
- with master commit: message write from client to server is truncated to 20, as is suppose to be according to this message ` E (10273) BT: attribute value too long, to be truncated to 20`
- server is receiving truncated message with length 20 byte

Logs:
- client logs:
```
D (301376) BLERemoteCharacteristic: >> writeValue(), length: 23
D (301376) FreeRTOS: Semaphore taking: name: WriteCharEvt (0x0), owner: <N/A> for writeValue
D (301383) FreeRTOS: Semaphore taken:  name: WriteCharEvt (0x0), owner: writeValue
E (301391) BT: attribute value too long, to be truncated to 20
V (301392) FreeRTOS: >> wait: Semaphore waiting: name: WriteCharEvt (0x0), owner: writeValue for writeValue
D (301399) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_WRITE_CHAR_EVT
D (301416) BLEUtils: GATT Event: ESP_GATTC_WRITE_CHAR_EVT
D (301421) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, handle: 42 0x2a]
V (301428) FreeRTOS: Semaphore giving: name: WriteCharEvt (0x0), owner: writeValue
V (301436) FreeRTOS: << wait: Semaphore released: name: WriteCharEvt (0x0), owner: <N/A>
D (301444) BLERemoteCharacteristic: << writeValue
```
- server logs:
```
[D][BLECharacteristic.cpp:283] handleGATTServerEvent():  - Response to write event: New value: handle: 2a, uuid: 0d563a58-196a-48ce-ace2-dfec78acc814
[D][BLECharacteristic.cpp:286] handleGATTServerEvent():  - Data: length: 20, data: 54696d652073696e636520626f6f743a20393330
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_WRITE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_WRITE_EVT
[D][BLEUtils.cpp:1830] dumpGattServerEvent(): [conn_id: 0, trans_id: 278, bda: 30:ae:a4:06:4f:22, handle: 0x2a, offset: 0, need_rsp: 0, is_prep: 0, len: 20]
[D][BLEUtils.cpp:1832] dumpGattServerEvent(): [Data: 54696d652073696e636520626f6f743a20323832]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
[D][BLECharacteristic.cpp:654] setValue(): >> setValue: length=20, data=54696d652073696e636520626f6f743a20323832, characteristic UUID=0d563a58-196a-48ce-ace2-dfec78acc814
[D][BLECharacteristic.cpp:661] setValue(): << setValue
```
And now surprice. With the same client code but flashed from arduino-ide, which means its not the master commit we dont have issue with message truncating. Logs from server:
```
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
[D][BLEValue.cpp:44] addPart(): >> addPart: length=18
[D][BLECharacteristic.cpp:283] handleGATTServerEvent():  - Response to write event: New value: handle: 2a, uuid: 0d563a58-196a-48ce-ace2-dfec78acc814
[D][BLECharacteristic.cpp:286] handleGATTServerEvent():  - Data: length: 18, data: 54696d652073696e636520626f6f743a2033
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1782] dumpGattServerEvent(): [status: ESP_GATT_OK, handle: 0x2a]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_WRITE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_WRITE_EVT
[D][BLEUtils.cpp:1830] dumpGattServerEvent(): [conn_id: 0, trans_id: 69, bda: 30:ae:a4:06:4f:22, handle: 0x2a, offset: 18, need_rsp: 1, is_prep: 1, len: 7]
[D][BLEUtils.cpp:1832] dumpGattServerEvent(): [Data: 32333732303030]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
[D][BLEValue.cpp:44] addPart(): >> addPart: length=7
[D][BLECharacteristic.cpp:283] handleGATTServerEvent():  - Response to write event: New value: handle: 2a, uuid: 0d563a58-196a-48ce-ace2-dfec78acc814
[D][BLECharacteristic.cpp:286] handleGATTServerEvent():  - Data: length: 7, data: 32333732303030
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1782] dumpGattServerEvent(): [status: ESP_GATT_OK, handle: 0x2a]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_EXEC_WRITE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_EXEC_WRITE_EVT
[D][BLEUtils.cpp:1756] dumpGattServerEvent(): [conn_id: 0, trans_id: 70, bda: 30:ae:a4:06:4f:22, exec_write_flag: 0x01=WRITE]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_EXEC_WRITE_EVT
[D][BLEValue.cpp:66] commit(): >> commit

```

Yulong-espressif, 2017-12-27:
@chegewara The first question, when your MTU configured as 50, you should receive this ESP_GATTS_MTU_EVT event on the server side? If you receive this event, can you print the value of param-> mtu.mtu?  If you don't receive this event, The MTU will use the default value of 23, so the size of the notify is MTU-3 (MTU - 1 byte op_code + 2 byte handle) = 20 bytes.

The second question, from the log you provide, did not find any problems. When the client wants to send 25 bytes of data, because the MTU size is 23, so the Host stack will divide this data into two packets and use prepare write and execute write to send, use prepare write, when a data The packet can send up to bytes of MTU-5 (MTU - (1 byte op_code + 2 byte handle + 2 byte offset)), so when the server receives the first ESP_GATTS_WRITE_EVT event, length = 18 and param-> write .is_prep = true, length = 25 - 18 = 7 for the second ESP_GATTS_WRITE_EVT event, and param-> write.is_prep = true and param-> write.offset = 18. Upon completion of the data, you receive ESP_GATTS_EXEC_WRITE_EVT. If you still do not understand what, please refer to our esp-idf \ examples \ bluetooth \ gatt_server demo. Thank you very much.

chegewara, 2017-12-27:
@Yulong-espressif Yes, when i set mtu with nRF im receiving ESP_GATTS_MTU_EVT and im setting up this value in application to provide proper payload fragmentation before we send it.

I understand all of this, but like i said and what you can see on logs. With arduino-ide send message from ble client message as long as 25 bytes is divided and send in 2 packets (18+7 bytes). The same code and the same payload but with esp-idf sends just one packet truncated to 20 bytes, so we lost 5 bytes of payload. Server receives not complete payload.

Again, in first and second logging from this comment https://github.com/espressif/esp-idf/issues/1434#issuecomment-354006292:
- first logging shows that we want to send 23 bytes payload: `D (301376) BLERemoteCharacteristic: >> writeValue(), length: 23`
- also first logging shows that BT stack is truncating payload to 20 bytes: `E (301391) BT: attribute value too long, to be truncated to 20`
- from second loggings we can see that server is receiving only 20 bytes: `[D][BLECharacteristic.cpp:286] handleGATTServerEvent():  - Data: length: 20, data: 54696d652073696e636520626f6f743a20393330`
- and here: `[D][BLECharacteristic.cpp:654] setValue(): >> setValue: length=20, data=54696d652073696e636520626f6f743a20323832, characteristic UUID=0d563a58-196a-48ce-ace2-dfec78acc814`
so we received just part of our original payload.
Now the same client application with arduino-ide is providing fragmentation and sending 2 packets (18+7 bytes):
- `[D][BLECharacteristic.cpp:286] handleGATTServerEvent():  - Data: length: 18, data: 54696d652073696e636520626f6f743a2033`
- `[D][BLEUtils.cpp:1832] dumpGattServerEvent(): [Data: 32333732303030]`.

Conclusion:
esp-idf master commit does not provide fragmentation for long payload but is truncating it to 20 bytes.

chegewara, 2017-12-27:
Also from this comment https://github.com/espressif/esp-idf/issues/1434#issuecomment-353998807 we can see that BT stack is truncating notification payload to 20 bytes. Even if we change MTU to 50 bytes, we have message `[D][BLECharacteristic.cpp:535] notify(): - Truncating to 20 bytes (maximum notify size)
` and client is receiving only 20 bytes, but bluetooth specifications says that notification payload can be larger than 20 bytes when we change MTU:
`According to the Bluetooth 4.0 specification Vol 3 Part F 3.4.7.1"If the attribute value is longer than (ATT_MTU–3) octets, then only the first (ATT_MTU–3) octets of this attributes value can be sent in a notification."`

negativekelvin, 2017-12-27:
`[D][BLECharacteristic.cpp:535] notify(): - Truncating to 20 bytes (maximum notify size) `

This is from nkolban code not bluedroid

chegewara, 2017-12-27:
Thanks, i dont know how did i miss it. But how about difference between master commit and arduino-ide?
Arduino-ide seems works fine but master commit is truncating messages. And im not saying now about notify `[D][BLECharacteristic.cpp:535] notify(): - Truncating to 20 bytes (maximum notify size)` but this one `E (301391) BT: attribute value too long, to be truncated to 20`

Yulong-espressif, 2017-12-27:
@chegewara Please look care of the log, ESP_GATTS_WRITE_EVT this type of event name of course from IDF.

chegewara, 2017-12-27:
@Yulong-espressif If there is something i am missing please tell me. Im not master at BLE more like apprentice. I can see that with arduino-ide it works and with esp-idf its not by using the same client and server code and serial console and nRF connect to confirm value written to server. I see issue with` [D][BLECharacteristic.cpp:535] notify(): - Truncating to 20 bytes (maximum notify size)`. Its bug in BLE library and we will fix it easy, but i cant find anything about ESP_GATTS_WRITE_EVT except that BT is truncating payload to 20 bytes.
Im sorry if im missing something obvious, im trying to help not make anyone angry at me.

chegewara, 2017-12-27:
This is the original issue that we came from 
https://github.com/nkolban/esp32-snippets/issues/316

Yulong-espressif, 2017-12-27:
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
[D][BLEValue.cpp:44] addPart(): >> addPart: length=18
[D][BLECharacteristic.cpp:283] handleGATTServerEvent():  - Response to write event: New value: handle: 2a, uuid: 0d563a58-196a-48ce-ace2-dfec78acc814
[D][BLECharacteristic.cpp:286] handleGATTServerEvent():  - Data: length: 18, data: 54696d652073696e636520626f6f743a2033
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1782] dumpGattServerEvent(): [status: ESP_GATT_OK, handle: 0x2a]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_WRITE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_WRITE_EVT
[D][BLEUtils.cpp:1830] dumpGattServerEvent(): [conn_id: 0, trans_id: 69, bda: 30:ae:a4:06:4f:22, handle: 0x2a, offset: 18, need_rsp: 1, is_prep: 1, len: 7]
[D][BLEUtils.cpp:1832] dumpGattServerEvent(): [Data: 32333732303030]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
[D][BLEValue.cpp:44] addPart(): >> addPart: length=7
[D][BLECharacteristic.cpp:283] handleGATTServerEvent():  - Response to write event: New value: handle: 2a, uuid: 0d563a58-196a-48ce-ace2-dfec78acc814
[D][BLECharacteristic.cpp:286] handleGATTServerEvent():  - Data: length: 7, data: 32333732303030
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1782] dumpGattServerEvent(): [status: ESP_GATT_OK, handle: 0x2a]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_EXEC_WRITE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_EXEC_WRITE_EVT
[D][BLEUtils.cpp:1756] dumpGattServerEvent(): [conn_id: 0, trans_id: 70, bda: 30:ae:a4:06:4f:22, exec_write_flag: 0x01=WRITE]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_EXEC_WRITE_EVT
[D][BLEValue.cpp:66] commit(): >> commit

chegewara, 2017-12-27:
@Yulong-espressif Yes, its from BLE client compiled with arduino-ide and this is what we expecting and works good.

chegewara, 2017-12-27:
This is from esp-idf compilation:
```
D (301376) BLERemoteCharacteristic: >> writeValue(), length: 23
D (301376) FreeRTOS: Semaphore taking: name: WriteCharEvt (0x0), owner: <N/A> for writeValue
D (301383) FreeRTOS: Semaphore taken:  name: WriteCharEvt (0x0), owner: writeValue
E (301391) BT: attribute value too long, to be truncated to 20
V (301392) FreeRTOS: >> wait: Semaphore waiting: name: WriteCharEvt (0x0), owner: writeValue for writeValue
D (301399) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_WRITE_CHAR_EVT
D (301416) BLEUtils: GATT Event: ESP_GATTC_WRITE_CHAR_EVT
D (301421) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, handle: 42 0x2a]
V (301428) FreeRTOS: Semaphore giving: name: WriteCharEvt (0x0), owner: writeValue
V (301436) FreeRTOS: << wait: Semaphore released: name: WriteCharEvt (0x0), owner: <N/A>
D (301444) BLERemoteCharacteristic: << writeValue
```

negativekelvin, 2017-12-27:
Error comes from here but can't say why
https://github.com/espressif/esp-idf/blob/a4fe12cb6d195a580a57690ceed45e1e06dc58e0/components/bt/bluedroid/stack/gatt/att_protocol.c#L317-L325

Yulong-espressif, 2017-12-27:
@chegewara In this log, I have seen the BT stack of IDF have receive two write event（ESP_GATTS_WRITE_EVT） and one exec write event （ESP_GATTS_EXEC_WRITE_EVT） in the server side. The frist write event length = 18, and the second write event length = 7. So what is the problem? Do not worry， I just don't know how to get your real idea.

Yulong-espressif, 2017-12-27:
@negativekelvin This error log just to remind the user the data will be fragment by the BT stack automatic.

chegewara, 2017-12-27:
@Yulong-espressif Just forget that log. This is very good log, application works fine, there is no issue. Its from arduino-ide, and as you said we can see 2 write_evt (one with 18 bytes and one with 7 bytes PDU). This is what im expecting to see when i compile and flash the same code with esp-idf:
```
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
[D][BLEValue.cpp:44] addPart(): >> addPart: length=18
[D][BLECharacteristic.cpp:283] handleGATTServerEvent(): - Response to write event: New value: handle: 2a, uuid: 0d563a58-196a-48ce-ace2-dfec78acc814
[D][BLECharacteristic.cpp:286] handleGATTServerEvent(): - Data: length: 18, data: 54696d652073696e636520626f6f743a2033
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1782] dumpGattServerEvent(): [status: ESP_GATT_OK, handle: 0x2a]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_WRITE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_WRITE_EVT
[D][BLEUtils.cpp:1830] dumpGattServerEvent(): [conn_id: 0, trans_id: 69, bda: 30:ae:a4:06:4f:22, handle: 0x2a, offset: 18, need_rsp: 1, is_prep: 1, len: 7]
[D][BLEUtils.cpp:1832] dumpGattServerEvent(): [Data: 32333732303030]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
[D][BLEValue.cpp:44] addPart(): >> addPart: length=7
[D][BLECharacteristic.cpp:283] handleGATTServerEvent(): - Response to write event: New value: handle: 2a, uuid: 0d563a58-196a-48ce-ace2-dfec78acc814
[D][BLECharacteristic.cpp:286] handleGATTServerEvent(): - Data: length: 7, data: 32333732303030
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1782] dumpGattServerEvent(): [status: ESP_GATT_OK, handle: 0x2a]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:94] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_EXEC_WRITE_EVT
[D][BLEUtils.cpp:1645] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_EXEC_WRITE_EVT
[D][BLEUtils.cpp:1756] dumpGattServerEvent(): [conn_id: 0, trans_id: 70, bda: 30:ae:a4:06:4f:22, exec_write_flag: 0x01=WRITE]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_EXEC_WRITE_EVT
[D][BLEValue.cpp:66] commit(): >> commit
```

chegewara, 2017-12-27:
But im having issue with this logs. Just one write_evt, PDU is truncated by BT and this is compiled and flashed with esp-idf master commit (unlike arduino-ide):
```
D (301376) BLERemoteCharacteristic: >> writeValue(), length: 23
D (301376) FreeRTOS: Semaphore taking: name: WriteCharEvt (0x0), owner: <N/A> for writeValue
D (301383) FreeRTOS: Semaphore taken:  name: WriteCharEvt (0x0), owner: writeValue
E (301391) BT: attribute value too long, to be truncated to 20
V (301392) FreeRTOS: >> wait: Semaphore waiting: name: WriteCharEvt (0x0), owner: writeValue for writeValue
D (301399) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_WRITE_CHAR_EVT
D (301416) BLEUtils: GATT Event: ESP_GATTC_WRITE_CHAR_EVT
D (301421) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, handle: 42 0x2a]
V (301428) FreeRTOS: Semaphore giving: name: WriteCharEvt (0x0), owner: writeValue
V (301436) FreeRTOS: << wait: Semaphore released: name: WriteCharEvt (0x0), owner: <N/A>
D (301444) BLERemoteCharacteristic: << writeValue
```

chegewara, 2017-12-27:
If fragmenting PDU is something that we should have done in library then we will do, but its not i understand from Your first response to mr Kolban:
https://github.com/espressif/esp-idf/issues/1434#issuecomment-353906019

As far as i understand BT stack is taking care of of fragmenting too long PDU before sending it with write_evt.

chegewara, 2017-12-27:
Here is longer log to show that there is no fragmenting and only one write_evt occurs:
```
D (17262779) BLERemoteCharacteristic: >> writeValue(), length: 22
D (17262779) FreeRTOS: Semaphore taking: name: WriteCharEvt (0x3ffce1d8), owner: <N/A> for writeValue
D (17262788) FreeRTOS: Semaphore taken:  name: WriteCharEvt (0x3ffce1d8), owner: writeValue
E (17262796) BT: attribute value too long, to be truncated to 20
V (17262797) FreeRTOS: >> wait: Semaphore waiting: name: WriteCharEvt (0x3ffce1d8), owner: writeValue for writeValue
D (17262804) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_WRITE_CHAR_EVT
D (17262822) BLEUtils: GATT Event: ESP_GATTC_WRITE_CHAR_EVT
D (17262827) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, handle: 42 0x2a]
V (17262834) FreeRTOS: Semaphore giving: name: WriteCharEvt (0x3ffce1d8), owner: writeValue
V (17262843) FreeRTOS: << wait: Semaphore released: name: WriteCharEvt (0x3ffce1d8), owner: <N/A>
D (17262852) BLERemoteCharacteristic: << writeValue
D (17263857) SampleClient: Setting the new value
D (17263857) BLERemoteCharacteristic: >> writeValue(), length: 22
D (17263857) FreeRTOS: Semaphore taking: name: WriteCharEvt (0x3ffce1d8), owner: <N/A> for writeValue
D (17263866) FreeRTOS: Semaphore taken:  name: WriteCharEvt (0x3ffce1d8), owner: writeValue
E (17263874) BT: attribute value too long, to be truncated to 20
V (17263875) FreeRTOS: >> wait: Semaphore waiting: name: WriteCharEvt (0x3ffce1d8), owner: writeValue for writeValue
D (17263882) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_WRITE_CHAR_EVT
D (17263900) BLEUtils: GATT Event: ESP_GATTC_WRITE_CHAR_EVT
D (17263905) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, handle: 42 0x2a]
V (17263912) FreeRTOS: Semaphore giving: name: WriteCharEvt (0x3ffce1d8), owner: writeValue
V (17263921) FreeRTOS: << wait: Semaphore released: name: WriteCharEvt (0x3ffce1d8), owner: <N/A>
D (17263930) BLERemoteCharacteristic: << writeValue
D (17264935) SampleClient: Setting the new value
```

Yulong-espressif, 2017-12-27:
@chegewara Can you show the the problem log code to me?

chegewara, 2017-12-27:
```
D (17263857) BLERemoteCharacteristic: >> writeValue(), length: 22  <-- NO MATTER HOW MUCH ITS OVER 20 BYTES
E (17262796) BT: attribute value too long, to be truncated to 20  <-- I UNDERSTAND ITS JUST LOG MESSAGE AND NOT REAL ERROR
D (17262804) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_WRITE_CHAR_EVT <-- JUST ONE OCCURENCE
```
On peripheral (esp32) is installed server application and is receiving ONLY 20 bytes, there is no prep_write or set flag is_long=1

chegewara, 2017-12-27:
@Yulong-espressif i will prepare test app just with esp-idf, without BLE library from Neil's Kolban and be back later.

chegewara, 2017-12-28:
@Yulong-espressif I have to admit that tests with esp-idf example server and client tells me i have made something worng with my own application tests. PDU written from client is properly segmented and received by server. Thank you for your time.

closed 2018-01-11 by FayeY