espressif/esp-idf #1435: Get assertion in start_cpu0_default()

HuyK, 2017-12-26:
Hi all,

I am using ESP32S module for my project. I made two boards which have ESP32S modules sitting on them
One board can run debug in Eclipse normally. The other board, I always get this assertion:
  assert(rtc_clk_xtal_freq_get() != RTC_XTAL_FREQ_AUTO);
in esp_clk_init() which is called by start_cpu0_default() function.
At that assertion, there is a comment about upgrading bootloader or enable CONFIG_COMPATIBLE_PRE_V2_1_BOOTLOADERS if the assertion is failed.

Can anyone help to know the reason why one board always gets that assertion? How to solve it?

Thanks
HuyK

Spritetm, 2017-12-26:
My guess would be that the bootloader in one board is old, and Eclipse only reflashes the app. See if you can reprogram everything (normally done using 'make flash', I don't know how it works under Eclipse), that should update the bootloader as well.

HuyK, 2017-12-26:
yeah, I use "make flash" to program both bootloader and application. Now it starts to work fine.
Thank you

closed 2017-12-26 by HuyK