espressif/esp-idf #1436: Feature Request: FreeRTOS 10 -- Stream Buffers & Message Buffers

hchaudhary1, 2017-12-26:
ESP-IDF team:
Is it possible to backport these amazing new features of FreeRTOS 10?

https://www.freertos.org/RTOS-stream-buffer-API.html
https://www.freertos.org/RTOS-message-buffer-API.html

RichardBarry, 2017-12-27:
You may be able to just include the stream_buffer.c and stream_buffer.h files from V10 in a V9 project.  That will give you the API.  I think the only dependency is Direct To Task Notifications, were are also available in V9 (assuming that it the base version you are wishing to back port to)..

hchaudhary1, 2017-12-28:
Do I need to allocate any functions to the IRAM, or is executing these kernel level functions flash ok?

hchaudhary1, 2017-12-28:
does not work..
looks like notify functionality is missing in the latest IDF.. base FreeRTOS version is v8

hchaudhary1, 2017-12-29:
got it to work... pull request added #1447 
https://github.com/espressif/esp-idf/pull/1447

closed 2018-02-20 by hchaudhary1