espressif/esp-idf #1437: fread returns 0 instead of number of bytes read.

muhgin, 2017-12-27:
I tested with spiffs and fat, both fread returned 0 in master repo. 
It was working before.

igrr, 2017-12-27:
Could you please attach some code which demonstrates the issue? fread seems
to work correctly both in our unit tests, and sd_card example, if fgets is
replaced by fread.

muhgin, 2017-12-27:
I'm sorry I have to amend.

My issue is that when i flash spiffs image using mkspiffs, when I tried to read the file(after successfully opened it), the function returned -1.
e.g:
 	fd = open(path, O_RDONLY);
	if (fd < 0)
		return;
	readLen = read(fd, buf, FILE_LEN); //readLen returned -1.

It doesn't happen if I create the file in the code(instead of flashing spiffs image), like sd_card example.

igrr, 2017-12-27:
It might be worth checking if mkspiffs is compiled with the same config as spiffs library for the ESP32. Being able to see the file but not read its contents usually is an indicator of configuration mismatch.

muhgin, 2017-12-27:
Thanks for the hint but I think I got the configuration right, as it has worked previously.
I might missing something though, but below is my configuration.

e.g: 
1) My partition table, spiffs at 0x110000
***********************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
storage,data,spiffs,0x110000,1M,
***********************************************

2) Mkspiffs command, Block Size: 4096, Page Size: 256 
**./mkspiffs -c webserver -b 4096 -p 256 -s 0x100000 spiffs.bin**

3) image flashing command at 0x110000
**esptool.py --chip esp32 --port COM5 --baud 1500000 write_flash -z 0x110000 spiffs.bin**

igrr, 2017-12-27:
Previously in IDF `OBJ_META_LEN` was set to zero by default, now it is set to 4 by default:
https://github.com/espressif/esp-idf/blob/master/components/spiffs/include/spiffs_config.h#L161

In IDF, you can modify this option via menuconfig: http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#spiffs-meta-length

In mkspiffs, `OBJ_META_LEN` is set to zero: https://github.com/igrr/mkspiffs/blob/master/include/spiffs_config.h#L140

So if you want to load an image created using mkspiffs, `CONFIG_SPIFFS_META_LENGTH` must be set to zero in menuconfig.

muhgin, 2017-12-27:
Yes, that solved it. Thanks a lot igrr.

you guys are inspiring, i really need to read a lot more than i imagined.

igrr, 2017-12-27:
Good to know that this fixed the issue for you. We'll update ESP-IDF documentation to mention this point.

JohnRabbit54, 2018-02-01:
The problem is that this is not only an issue with mkspiff. Files, in old filesystems, made with SPIFFS_META_LENGTH=0, are accessible but unreadable. Unless you made SPIFFS_META_LENGTH=0. Thats fine in a ESP-IDF environment and ESP32-Arduino as a component in ESP-IDF. However in a "native" Arduino environment with ESP32-Arduino (from GitHub) did doesn't work because you cann't modify the SPIFFS_META_LENGTH.

igrr, 2018-02-12:
mkspiffs has been updated, now there is a release version with configuration matching that of ESP-IDF: https://github.com/igrr/mkspiffs/releases (look for mkspiffs-0.2.2-esp-idf-YOUR_OS archive)

closed 2018-02-12 by igrr