espressif/esp-idf #1439: rmt_write_items() doesn't give semaphore when wait_tx_done is false

t-mat, 2017-12-27:
When I call `rmt_*()` functions in the following order, `rmt_driver_uninstall()` is locked up.

```C
rmt_config(_);
rmt_driver_install(chan, 0, 0);
rmt_set_tx_intr_en(chan, 0);
rmt_write_items(chan, _, _, false);
rmt_tx_stop();
rmt_driver_uninstall(chan);             // <---
```

The reason is `rmt_write_items()` just takes and doesn't give semaphore when `wait_tx_done` (last argument) is `false`.

```C
esp_err_t rmt_write_items(rmt_channel_t channel, const rmt_item32_t* rmt_item, int item_num,
                          bool wait_tx_done)
{
    ...
    xSemaphoreTake(p_rmt->tx_sem, portMAX_DELAY);
    ...
    rmt_tx_start(channel, true);
    if(wait_tx_done) {                                 // <---
        xSemaphoreTake(p_rmt->tx_sem, portMAX_DELAY);
        xSemaphoreGive(p_rmt->tx_sem);                 // <---
    }
    return ESP_OK;
}
```

And since no other functions don't give semaphore, `rmt_driver_uninstall()` is locked up at this line.

```C
esp_err_t rmt_driver_uninstall(rmt_channel_t channel)
{
    ...
    xSemaphoreTake(p_rmt_obj[channel]->tx_sem, portMAX_DELAY);  // <---
    ...
}
```

Possible resolutions are

(1) `rmt_tx_stop()` give semaphore when it confirm TX is stopped.
(2) Add "give semaphore" in `rmt_write_items()` if `wait_tx_done` is false.
(3) `rmt_driver_uninstall()` ignores semaphore.

I'd like to use method (1).  Because (2) and (3) may introduce other problems.

Possible patch may seem like this:

```C
esp_err_t rmt_tx_stop(rmt_channel_t channel)
{
    RMT_CHECK(channel < RMT_CHANNEL_MAX, RMT_CHANNEL_ERROR_STR, ESP_ERR_INVALID_ARG);
    portENTER_CRITICAL(&rmt_spinlock);
    RMT.conf_ch[channel].conf1.tx_start = 0;
    portEXIT_CRITICAL(&rmt_spinlock);
    xSemaphoreGive(p_rmt_obj[channel]->tx_sem);   ///+
    return ESP_OK;
}
```

koobest, 2018-01-03:
Hi, @t-mat,
yes, you are write, this is really a bug, we are working on it. thanks a lot.

t-mat, 2018-01-22:
@koobest , > Hi, @t-mat, Is it necessary to turn off the interrupt in your application?

Yes, it is.  Because ESP-IDF always forces users to use `rmt_driver_install(chan,...)` and `rmt_driver_uninstall(chan)` pair for each RMT channel which is occupied by user.
If I want to reuse same RMT channel for other purpose in the different settings, I must call `rmt_driver_uninstall(chan)`.

> to some extent, I think, it is abnormal to use it lik that.

I don't think so.  Honestly, I don't understand why you use the word "abnormal" here.
Please note that [`rmt_driver_uninstall(chan)` is explicitly documented](https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/rmt.html#uninstall-driver).  And I've never seen any API set which prohibit to call deallocation function even though it has complete allocation/deallocation API pair.

What do you think about "normal" usage of RMT APIs?

koobest, 2018-01-23:
@t-mat, I'm sorry for my unclear expression. my mean is that disable all interruption before write_item will certainly lead to driver_uninstall block. Can you disable the interrupt after rmt_tx_stop ?

koobest, 2018-01-29:
Hi, @t-mat 
Has your problem been solved?

t-mat, 2018-01-29:
> Has your problem been solved?

No.  My problem and this issue are not resolved.

> my mean is that disable all interruption before write_item will certainly lead to driver_uninstall block. Can you disable the interrupt after rmt_tx_stop ?

My example code might not illustrate my problem correctly.  I think the following code may express the issue

```
rmt_config_t c = { ... };
c.tx_config.tx.loop_en = 1;
rmt_config(&c);
rmt_driver_install(chan, 0, 0);
rmt_set_tx_intr_en(chan, 0);
rmt_write_items(chan, _, _, false);    // Last argument (wait_tx_done) = false

... do something ...

auto& rmtConf = RMT.conf_ch[chan].conf1; // *
rmtConf.tx_start = 1;                    // * Send RMT

... do something ...

rmt_tx_stop();
rmt_driver_uninstall(chan);             // <--- stall
```

I didn't use `rmt_write_items()` to activate TX.  Since I want to start TX in the tight timing, I use RMT register directly (`*` comment lines).
After end of TX, I want to release the RMT channel by `rmt_driver_uninstall()` but it've stalled.

I didn't address to any ISRs in the RMT component.  esp-idf requires to call `rmt_driver_uninstall()` and I just follow the requirement.  I don't care about that RMT component disables its ISR or not.

koobest, 2018-02-06:
Hi, @t-mat 
    we add a  bool variable  `wait_done`   in ` rmt_obj_t`  and change its value in `rmt_write_items`. in  `rmt_driver_uninstall` we check the valure of  `wait_done`, if it not true, we do not wait for the semaphore.  but I'm not sure if it works in your application.

closed 2018-10-30 by Alvin1Zhang