espressif/esp-idf #1440: i2c fails to read data as master

FlavioBayer, 2017-12-27:
Commit d78be97154590ae5d48ebbb900182711c035274e has introduced a check in order to assert the number of bytes to read when acting as master (https://github.com/espressif/esp-idf/blob/d78be97154590ae5d48ebbb900182711c035274e/components/driver/i2c.c#L1008).

This assertion only consider length less than 1 (0 bytes to read) as valid, which obviosly makes no sense and is not the expected behavior.
If something, it should assert data_len>0, maybe data_len>=0 (which would make no sense since it is unsigned anyways).

PerMalmberg, 2017-12-29:
There is already a PR for this, merge pending.. https://github.com/espressif/esp-idf/pull/1443

projectgus, 2018-01-02:
Fixed in master now (backport to v3.0 branch coming soon), sorry for the inconvenience.

closed 2018-01-02 by projectgus