espressif/esp-idf #1442: BLE authentication question

chegewara, 2017-12-28:
Is it possible to use hardcoded passkey PIN to perform authentication?

chegewara, 2017-12-30:
@Yulong-espressif Any help with this, please.

Yulong-espressif, 2017-12-30:
@chegewara I don't understand what is hardcoded passkey PIN? Can you explain?

Yulong-espressif, 2017-12-30:
@chegewara  And  what is the difference with ordinary passkey entry?

chegewara, 2017-12-30:
As for now bluedroid is generating random passkey during authentication. Its random 6 digit number. Ive been asked if its possible to have hardcoded by app creator PIN, that will be a part of user code and always the same for device or app. Im suspecting that user wants to have authentication with PIN, not ordinary just_works, but his device does not provide option to display passkey

chegewara, 2017-12-30:
I thought it could be done with OOB, but as far as i understant OOB is working with NFC etc. In mesh networking it would works with OOB, because there is option to have passkey on piece of paper.

chegewara, 2017-12-30:
@Yulong-espressif 
I thought it could be done with OOB, but as far as i understant OOB is working with NFC etc. In mesh networking it would works with OOB, because there is option to have passkey on piece of paper.

Yulong-espressif, 2017-12-30:
@chegewara We don't support the OOB method now, we will support it in the next version(v3.1). Thanks.

chegewara, 2017-12-30:
@Yulong-espressif I dont support it either, im not as good. How about that hardcoded PIN? Is it possible, or user will have to use just_works authentication with encrypted connection?

Yulong-espressif, 2017-12-30:
@chegewara We also still don't support hard key method now, I think it is a useful method, we will support it recently. Please continue to focus on the IDF. Thank you.

chegewara, 2017-12-30:
@Yulong-espressif Thank you and have a nice day

Yulong-espressif, 2017-12-30:
@chegewara It's my honor. An early Happy new year to you.

chegewara, 2018-01-03:
I have another question about SMP this time. I want to switch off part of my code related security layer if user turn off SMP in menuconfig (`CONFIG_BLE_SMP_ENABLE=n`). Now is the question, as far as i understand functions like `esp_ble_set_encryption()` or `esp_ble_passkey_reply()` for example are part of SMP layer but definitions can be found in esp_gap_ble_api.h, is that correct? If i am right then `CONFIG_BLE_SMP_ENABLE=n` does nothing because this is only this assertion `ESP_BLUEDROID_STATUS_CHECK(ESP_BLUEDROID_STATUS_ENABLED);`.

chegewara, 2018-01-05:
@Yulong-espressif any help with this?

Yulong-espressif, 2018-01-05:
@chegewara Hi, sorry for the time now to give you a reply, it's like this, When you close the SMP in make menuconfig, all the BLE security-related functions will be shut down. The purpose of this is that When SMP functionality is not needed, we can save more RAM and code size. So when you disable the SMP function, esp_ble_set_encryption and esp_ble_passkey_reply related to the encryption function can not be used, the esp_gap_ble_api.c file has been added #if (SMP_INCLUDED == TRUE) macro, the encryption-related functions are carried out. So there is no need to isolate it in the esp_gap_ble_api.h header file because declarations in esp_gap_ble_api.h do not work when the compiler can not find a definition for it. So, if you don't enable the SMP macro in the menuconfig, when you used the BLE security-related functions there will be compilation errors.

chegewara, 2018-01-05:
I see it now, i dont know ho i missed this macro `#if (SMP_INCLUDED == TRUE) ` in esp_gap_ble_api.c. Thanks again

closed 2018-01-05 by chegewara