espressif/esp-idf #1449: Configured boot partition is not preserved when deep sleep is started as another task

ljozsa, 2017-12-29:
Hi,

is the issue from the title intentional or should I always explicitly set boot partition before entering the deep sleep?

Thanks,
L.

ljozsa, 2017-12-29:
I might extend my description a little bit. When the code is running from an ota partition and there is another task under app_main running that ends with entering the deep sleep, the next boot is not performed from that ota partition, instead the CPU boots from default partition.

My current workaround is to run 
`
        err = esp_ota_set_boot_partition(esp_ota_get_running_partition());
         if (err != ESP_OK) {
                ESP_LOGE(TAG, "esp_ota_set_boot_partition failed! err=0x%x", err);
                task_fatal_error();
            }
`
before entering deep sleep.

negativekelvin, 2017-12-30:
Can you show the bootloader logs?

ljozsa, 2017-12-30:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5544
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:13716
entry 0x40078fb4
I (30) boot: ESP-IDF v3.1-dev-107-g5401a75b 2nd stage bootloader
I (30) boot: compile time 10:15:53
I (30) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (67) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (75) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  3 factory          factory app      00 00 00010000 00100000
I (89) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (97) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (104) boot: End of partition table
I (109) esp_image: segment 0: paddr=0x00110020 vaddr=0x3f400020 size=0x118a4 ( 71844) map
I (143) esp_image: segment 1: paddr=0x001218cc vaddr=0x3ffb0000 size=0x03154 ( 12628) load
I (148) esp_image: segment 2: paddr=0x00124a28 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/ljozsa/esp/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (150) esp_image: segment 3: paddr=0x00124e30 vaddr=0x40080400 size=0x0b1e0 ( 45536) load
I (177) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0x54d78 (347512) map
0x400d0018: _stext at ??:?

I (298) esp_image: segment 5: paddr=0x00184d98 vaddr=0x4008b5e0 size=0x05a14 ( 23060) load
0x4008b5e0: set_tx_gain_table_bt at ??:?

I (308) esp_image: segment 6: paddr=0x0018a7b4 vaddr=0x400c0000 size=0x00064 (   100) load
I (318) boot: Loaded app from partition at offset 0x110000
I (318) boot: Disabling RNG early entropy source...
I (319) cpu_start: Pro cpu up.
I (322) cpu_start: Starting app cpu, entry point is 0x4008122c
0x4008122c: call_start_cpu1 at /home/ljozsa/esp/esp-idf/components/esp32/./cpu_start.c:222

I (314) cpu_start: App cpu up.
I (333) heap_init: Initializing. RAM available for dynamic allocation:
I (340) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (346) heap_init: At 3FFB9488 len 00026B78 (154 KiB): DRAM
I (352) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (358) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (365) heap_init: At 40090FF4 len 0000F00C (60 KiB): IRAM
I (371) cpu_start: Pro cpu start user code
I (54) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (120) wifi: wifi firmware version: 275023b
I (120) wifi: config NVS flash: enabled
I (120) wifi: config nano formating: disabled
I (120) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (130) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (160) wifi: Init dynamic tx buffer num: 32
I (160) wifi: Init data frame dynamic rx buffer num: 32
I (160) wifi: Init management frame dynamic rx buffer num: 32
I (170) wifi: wifi driver task: 3ffc083c, prio:23, stack:4096
I (170) wifi: Init static rx buffer num: 10
I (170) wifi: Init dynamic rx buffer num: 32
I (180) wifi: wifi power manager task: 0x3ffc547c prio: 21 stack: 2560
I (180) ota: Setting WiFi configuration SSID OpenWrt...
I (220) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0
I (220) wifi: mode : sta (30:ae:a4:03:0b:a4)
I (220) ota: Update server task started 
[... 174 lines cut ...]
I (176) wifi: wifi driver task: 3ffc083c, prio:23, stack:4096
I (176) wifi: Init static rx buffer num: 10
I (176) wifi: Init dynamic rx buffer num: 32
I (186) wifi: wifi power manager task: 0x3ffc547c prio: 21 stack: 2560
I (186) ota: Setting WiFi configuration SSID OpenWrt...
I (206) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 1
I (206) wifi: mode : sta (30:ae:a4:03:0b:a4)
I (206) ota: Update server task started 

I (916) wifi: n:6 1, o:1 0, ap:255 255, sta:6 1, prof:1
I (1896) wifi: state: init -> auth (b0)
I (1906) wifi: state: auth -> assoc (0)
I (1906) wifi: state: assoc -> run (10)
I (1936) wifi: connected with OpenWrt, channel 6
I (2586) event: sta ip: 192.168.71.198, mask: 255.255.255.0, gw: 192.168.71.1
I (2586) ota: Running partition type 0 subtype 0 (offset 0x00010000)
I (2586) ota: Writing to partition subtype 16 at offset 0x110000
I (2596) ota: esp_ota_begin succeeded
I (2596) ota: ... Allocated socket

I (2596) ota: ... Socket bind done 

I (4126) wifi: pm start, type:0

I (10226) deepsleep: Going to enter deep sleep now

chegewara, 2017-12-30:
`I (109) boot: Defaulting to factory image`
Why?

ljozsa, 2017-12-30:
Is this question for me or esp-idf devels? :-) I haven't touched the boot code. I just slightly modified ota example provided with the esp-idf SDK.

chegewara, 2017-12-30:
Yes, its question to devs

negativekelvin, 2017-12-30:
So if you hard reset it during deepsleep what happens?

ljozsa, 2017-12-30:
The behavior is the same as with waking up from deep sleep. Here's output with boot debugging on:

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7380
load:0x40078000,len:0
load:0x40078000,len:14820
entry 0x400792cc
I (29) boot: ESP-IDF v3.1-dev-107-g5401a75b 2nd stage bootloader
I (29) boot: compile time 13:28:52
I (29) boot: Enabling RNG early entropy source...
D (35) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
D (42) boot: magic e9
D (44) boot: segments 04
D (47) boot: spi_mode 02
D (49) boot: spi_speed 00
D (52) boot: spi_size 02
I (55) boot: SPI Speed      : 40MHz
I (59) boot: SPI Mode       : DIO
I (63) boot: SPI Flash Size : 4MB
D (67) bootloader_flash: mmu set paddr=00000000 count=1
D (72) boot: mapped partition table 0x8000 at 0x3f408000
D (77) flash_parts: partition table verified, 6 entries
I (83) boot: Partition Table:
I (86) boot: ## Label            Usage          Type ST Offset   Length
D (94) boot: load partition table entry 0x3f408000
D (99) boot: type=1 subtype=2
I (102) boot:  0 nvs              WiFi data        01 02 00009000 00004000
D (109) boot: load partition table entry 0x3f408020
D (114) boot: type=1 subtype=0
I (117) boot:  1 otadata          OTA data         01 00 0000d000 00002000
D (125) boot: load partition table entry 0x3f408040
D (130) boot: type=1 subtype=1
I (133) boot:  2 phy_init         RF data          01 01 0000f000 00001000
D (140) boot: load partition table entry 0x3f408060
D (145) boot: type=0 subtype=0
I (148) boot:  3 factory          factory app      00 00 00010000 00100000
D (156) boot: load partition table entry 0x3f408080
D (161) boot: type=0 subtype=10
I (164) boot:  4 ota_0            OTA app          00 10 00110000 00100000
D (172) boot: load partition table entry 0x3f4080a0
D (177) boot: type=0 subtype=11
I (180) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (187) boot: End of partition table
D (192) boot: OTA data offset 0xd000
D (195) bootloader_flash: mmu set paddr=00000000 count=1
D (201) boot: OTA sequence values A 0xffffffff B 0xffffffff
D (206) boot: OTA sequence numbers both empty (all-0xFF)
I (212) boot: Defaulting to factory image
D (216) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (223) esp_image: reading image header @ 0x10000
D (227) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (234) esp_image: image header: 0xe9 0x07 0x02 0x02 4008127c
I (240) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x11898 ( 71832) map
D (249) bootloader_flash: mmu set paddr=00010000 count=2
D (279) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (280) esp_image: segment 1: paddr=0x000218c0 vaddr=0x3ffb0000 size=0x03154 ( 12628) load
D (284) bootloader_flash: mmu set paddr=00020000 count=1
D (294) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (296) esp_image: segment 2: paddr=0x00024a1c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/ljozsa/esp/esp-idf/components/freertos/./xtensa_vectors.S:1685

D (305) bootloader_flash: mmu set paddr=00020000 count=1
D (311) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (317) esp_image: segment 3: paddr=0x00024e24 vaddr=0x40080400 size=0x0b1ec ( 45548) load
D (326) bootloader_flash: mmu set paddr=00020000 count=2
D (351) bootloader_flash: mmu set block paddr=0x00030000 (was 0xffffffff)
[... 894 lines cut ...]
I (443) wifi: Init dynamic rx buffer num: 32
I (443) wifi: wifi power manager task: 0x3ffc547c prio: 21 stack: 2560
I (453) ota: Setting WiFi configuration SSID OpenWrt...
I (493) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0
I (493) wifi: mode : sta (30:ae:a4:03:0b:a4)
I (493) ota: Update server task started 

I (1203) wifi: n:6 1, o:1 0, ap:255 255, sta:6 1, prof:1
I (2193) wifi: state: init -> auth (b0)
I (2193) wifi: state: auth -> assoc (0)
I (2203) wifi: state: assoc -> run (10)
I (2223) wifi: connected with OpenWrt, channel 6
I (2843) event: sta ip: 192.168.71.198, mask: 255.255.255.0, gw: 192.168.71.1
I (2843) ota: Running partition type 0 subtype 0 (offset 0x00010000)
I (2843) ota: Writing to partition subtype 16 at offset 0x110000
I (2863) ota: esp_ota_begin succeeded
I (2863) ota: ... Allocated socket

I (2863) ota: ... Socket bind done 

I (3123) wifi: pm start, type:0

I (10513) deepsleep: Going to enter deep sleep now

So right after flashing the new firmware over OTA, the code is booted from the correct partition but after it boots the configured partition reverted back to factory. However the only place where I'm calling esp_ota_set_boot_partition() is the code running immediately after ota transfer completes.

negativekelvin, 2017-12-30:
```
D (200) boot: OTA sequence values A 0x00000001 B 0xffffffff
D (206) boot: Only OTA sequence A is valid. Mapping seq 0 -> OTA slot 0
```

```
W (2851) ota: Configured OTA boot partition at offset 0x00010000, but running from offset 0x00110000
W (2851) ota: (This can happen if either the OTA boot data or preferred boot image become corrupted somehow.)
```

```
D (200) boot: OTA sequence values A 0xffffffff B 0xffffffff
D (206) boot: OTA sequence numbers both empty (all-0xFF)
I (211) boot: Defaulting to factory image
```

lennyJL, 2022-04-12:
Hi @ljozsa. Is your issue solved? If this issue persists, can you provide your code to help us reproduce this issue?

jack0c, 2022-04-14:
@ljozsa if you see the API `esp_ota_set_boot_partition` description, you should run `esp_ota_set_boot_partition` and  `esp_restart`, the deep sleep is just equal to `esp_restart`. So you are right, your workaround is necessary.
I suggest close this issue.

closed 2022-04-14 by Alvin1Zhang