espressif/esp-idf #1452: libcurl fails with "Connection already in progress"

paulreimer, 2017-12-30:
As noted in this issue:
https://github.com/nkolban/esp32-snippets/issues/219

I get a similar error with HTTPS using mbedTLS, as well:
```
* Rebuilt URL to: https://www.howsmyssl.com/
* timeout on name lookup is not supported
*   Trying 104.196.190.195...
* TCP_NODELAY set
* connect to 104.196.190.195 port 443 failed: Connection already in progress
* Failed to connect to www.howsmyssl.com port 443: Connection already in progress
* Closing connection 0
I (6156) curl_test: curl_easy_perform failed: Couldn't connect to server
```

Specifically these commits appear to break libcurl which worked before (e.g. on commit a320fed and before)
https://github.com/espressif/esp-idf/compare/a320fed...3f83914

I should note that libcurl does currently work on the latest release v2.1, but I ran `git bisect` to find the exact commits that broke it.

loboris, 2017-12-30:
My [ESP32 curl example](https://github.com/loboris/ESP32_curl_example) builds and runs with the latest esp-idf without any error.
It runs http/https GET/POST, FTP tests, SFTP tests, SMTP - Email test (GMail), ...

paulreimer, 2017-12-30:
Thanks @loboris, I can confirm that your `curl` component in the repo above does work for me using the latest `esp-idf` master (I can even use my `curl_config.h` that wasn't working above, but I have to comment out `SIZEOF_CURL_OFF_T`).

You also included `libssh2`, `quickmail`, and `zlib` to support more `curl` features, which I must say is awesome.

However I'm trying to figure out where you got the `curl` sources from. I see in your `include/curl/curlver.h`
```
#define LIBCURL_VERSION "7.54.1-DEV"
```

But there are significant diffs from the official `curl` v7.54.1 release sources. I'd prefer to be able to track upstream `curl`, at least know where the patches came from, etc.

loboris, 2017-12-31:
The libcurl sources were cloned from [curl repository](https://github.com/curl/curl).
Unfortunately, I don't remember the exact commit which was used.
I had to do some modifications to make it work with with ESP32.
I'll try to update the sources to the latest curl commit and document the needed changes when I find some time to do it.

loboris, 2017-12-31:
The problem with **Connection already in progress** is caused by the fact that the lwip sockets starts at `LWIP_SOCKET_OFFSET`.

`FD_SET`/`FD_GET` macros take this into account when operating on fd sets: `LWIP_SOCKET_OFFSET` is subtracted from fd number, and the resulting value as used as bit offset in fd set.

In curl (file _lib/select.h_), the macro `#define VALID_SOCK(s) (((s) >= 0) && ((s) < FD_SETSIZE))` checks the valid socket and **fails** with **esp-idf lwip** socket.

**_This macro has to be changed to_**:
`#define VALID_SOCK(s) (((s) - LWIP_SOCKET_OFFSET >= 0) && ((s) - LWIP_SOCKET_OFFSET < FD_SETSIZE))`,
then everything works as expected.

I've tested my [ESP32 curl example](https://github.com/loboris/ESP32_curl_example) with latest libcurl cloned from [curl repository](https://github.com/curl/curl) and it builds and runs without problem with this patch.

You can download all the **patches** I'm using from [my server](http://loboris.eu/ESP32/esp32_curl_paches.zip).

paulreimer, 2017-12-31:
@loboris Thank you! That macro change did resolve the issue for me, and I was also able to use the latest upstream `curl-7_57_0` (I already have a `curl_config.h`, and I do not need your `ssh` protocol patch at this time).

I never would have found where/how to modify that curl macro, either, so thanks again! My immediate concerns (for libcurl) are resolved.

However, as far as `esp-idf` goes, is this just an expected breaking change for 3.0? If it breaks `libcurl` (in quite a mysterious way), is it going to break other ported software, too?

Ideally there is a way to avoid this that can be added to `esp-idf`, but I am also curious in how you (@loboris) debugged this issue? I can kinda see that it would impact socket/select/`FD_` related things, and just hunt in the ported code for their usage (and re-adjust by the offset)?

loboris, 2018-01-01:
I think the libcurl (and other libraries using lwip) should be aware of lwip's `LWIP_SOCKET_OFFSET`.
As libcurl is lwip aware (`USE_LWIPSOCK`), the conditional compile should be used where `LWIP_SOCKET_OFFSET` can alter the code behavior.
This should be probably submited to curl repository as PR.

In esp-idf, LWIP is a special case for VFS use, and `LWIP_SOCKET_OFFSET` is set dynamically at VFS registration time. I've had some other issues since this behaviour was implemented, so it was easy to find the solution for curl issue.

projectgus, 2018-01-03:
> However, as far as esp-idf goes, is this just an expected breaking change for 3.0? If it breaks libcurl (in quite a mysterious way), is it going to break other ported software, too?

Good point. I've added a note to the release notes here:
https://github.com/espressif/esp-idf/releases/tag/v3.0-rc1
(Search for LWIP_SOCKET_OFFSET.)

This wasn't intended to be a breaking change (POSIX functions may return non-negative int as a valid fd), but it seems quite a lot of code assumes fds will always be low numbered integers (reasonable, as they usually are).

Some good news, LWIP_SOCKET_OFFSET will probably be going back to zero (or close to zero) in IDF v3.1 when we plan to expand select() to work with other VFS drivers. (Code updated for v3.0, like the modification for libcurl mentioned here, should continue to work.)

closed 2018-01-15 by FayeY

igrr, 2018-04-19:
Some news here, `LWIP_SOCKET_OFFSET` is not being used anymore. Lwip sockets once again use small numbers.

paulreimer, 2018-05-08:
I find that is not that case, at least with latest `master` I still get this error.

At least, I still require this patch in `libcurl`:
```
//#define VALID_SOCK(s) (((s) >= 0) && ((s) < FD_SETSIZE))
// Adjust the fd # by the LWIP_SOCKET_OFFSET to make it 0-based again
#define VALID_SOCK(s) (((s) - LWIP_SOCKET_OFFSET >= 0) && ((s) - LWIP_SOCKET_OFFSET < FD_SETSIZE))
```

igrr, 2018-05-08:
We had to revert VFS changes soon after they were done, due to a regression. Updated fix coming back soon.

paulreimer, 2018-08-23:
Safe to test this out yet?