espressif/esp-idf #1454: Using 3x OTA partitions without FACTORY partition

ammaree, 2017-12-30:
WE are trying to resolve the situation where a factory partition exist but the factory firmware becomes completely stale due to the inability/difficulty to upgrade the factory firmware.

Q: From what we understand the factory firmware can only be upgraded via UART, is this correct?

Q: If we create 3x OTA partitions, it seems we cannot get the OTA app flashed. How do we flash the first OTA to get the device going?

Q: Are there any limitations or inherent problems we should be aware of should we opt for a 3x OTA configuration without factory partition?

Thanks

AMM

chegewara, 2017-12-30:
Did you try to create 3x OTA partitions and flash first OTA partition from UART? When you create partitions table each partition has starting address, just try to flash first OTA partition. As far as i understand OTA partition is like standard factory partition with extra abilities.

ammaree, 2017-12-30:
Have created a new "partitions_three_ota.csv" file in esp-idf/components/partitions.
Rerun normal "make -j8 all flash monitor" command but return with "no rule to make target '.... partitions_three_ota.csv "
Cannot find any documentation explaining how to handle absence of "factory" partition.

chegewara, 2017-12-30:
Add file `partitions_three_ota.csv` to your project folder. Also you need to change setting with `make menuconfig` to Custom partitions table

ammaree, 2017-12-30:
I did make the change in make menuconfig but missed the fact that the CSV should be in project folder.
Working now, thanks...

loboris, 2017-12-31:
The **factory** partition **can** be updated (from OTA partition).
Just use:
`update_partition = esp_partition_find_first(ESP_PARTITION_TYPE_APP, ESP_PARTITION_SUBTYPE_APP_FACTORY, "part_name");`
instead of:
`update_partition = esp_ota_get_next_update_partition(NULL);`

ammaree, 2018-01-02:
Thanks, will try...

closed 2018-01-02 by ammaree