espressif/esp-idf #1456: Problem with RMT - the rmt_tx_stop command does not work

csann, 2017-12-31:
The rmt_tx_stop command does not work. It does not work if long series of pulses are generated by looping. It also does not work if short series of pulses are generated which does not require looping. The following examples are derived from the rmt_nec_tx_rx example in the peripherals folder of the esp-idf.

Example 1. Adding rmt_tx_stop command to 'infrared_nec_main.c'

`static void rmt_example_nec_tx_task()
{
    vTaskDelay(10);
    nec_tx_init();
    esp_log_level_set(NEC_TAG, ESP_LOG_INFO);
    int channel = RMT_TX_CHANNEL;
    uint16_t cmd = 0x0;
    uint16_t addr = 0x11;
    int nec_tx_num = RMT_TX_DATA_NUM;
    for(;;) {
        ESP_LOGI(NEC_TAG, "RMT TX DATA");
        size_t size = (sizeof(rmt_item32_t) * NEC_DATA_ITEM_NUM * nec_tx_num);
        //each item represent a cycle of waveform.
        rmt_item32_t* item = (rmt_item32_t*) malloc(size);
        int item_num = NEC_DATA_ITEM_NUM * nec_tx_num;
        memset((void*) item, 0, size);
        int i, offset = 0;
        while(1) {
            //To build a series of waveforms.
            i = nec_build_items(channel, item + offset, item_num - offset, ((~addr) << 8) | addr, cmd);
            if(i < 0) {
                break;
            }
            cmd++;
            addr++;
            offset += i;
        }
        //To send data according to the waveform items.
        rmt_write_items(channel, item, item_num, false);       // set to false
        
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        rmt_tx_stop(channel);                                                   // added
        
        //Wait until sending is done.
        //rmt_wait_tx_done(channel, portMAX_DELAY);         // removed
        //before we free the data, make sure sending is already done.
        free(item);
        vTaskDelay(2000 / portTICK_PERIOD_MS);
    }
    vTaskDelete(NULL);
}`

In the code above, the rmt_tx_stop command is ignored.

Example 2 Adding additional commands to 'infrared_nec_main.c'

`static void rmt_example_nec_tx_task()
{
    vTaskDelay(10);
    nec_tx_init();
    esp_log_level_set(NEC_TAG, ESP_LOG_INFO);
    int channel = RMT_TX_CHANNEL;
    uint16_t cmd = 0x0;
    uint16_t addr = 0x11;
    int nec_tx_num = RMT_TX_DATA_NUM;
    for(;;) {
        ESP_LOGI(NEC_TAG, "RMT TX DATA");
        size_t size = (sizeof(rmt_item32_t) * NEC_DATA_ITEM_NUM * nec_tx_num);
        //each item represent a cycle of waveform.
        rmt_item32_t* item = (rmt_item32_t*) malloc(size);
        int item_num = NEC_DATA_ITEM_NUM * nec_tx_num;
        memset((void*) item, 0, size);
        int i, offset = 0;
        while(1) {
            //To build a series of waveforms.
            i = nec_build_items(channel, item + offset, item_num - offset, ((~addr) << 8) | addr, cmd);
            if(i < 0) {
                break;
            }
            cmd++;
            addr++;
            offset += i;
        }
        //To send data according to the waveform items.
        rmt_write_items(channel, item, item_num, false);   // set to false
        
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        RMT.apb_conf.mem_tx_wrap_en = false;                // added
        rmt_tx_stop(channel);                                              // added
        rmt_set_tx_thr_intr_en(channel, 0, 0);                    // added
        
        //Wait until sending is done.
        //rmt_wait_tx_done(channel, portMAX_DELAY);         // removed
        //before we free the data, make sure sending is already done.
        free(item);
        vTaskDelay(2000 / portTICK_PERIOD_MS);
    }
    vTaskDelete(NULL);
}`

Adding the extra commands does cause transmission to stop however transmission does not continue when the next command is issued to the RMT. This is because of a semaphore problem in rmt.c rmt_write_items. A possible patch is this:

`esp_err_t rmt_write_items(rmt_channel_t channel, const rmt_item32_t* rmt_item, int item_num, bool wait_tx_done)
{
    RMT_CHECK(channel < RMT_CHANNEL_MAX, RMT_CHANNEL_ERROR_STR, ESP_ERR_INVALID_ARG);
    RMT_CHECK(p_rmt_obj[channel] != NULL, RMT_DRIVER_ERROR_STR, ESP_FAIL);
    RMT_CHECK(rmt_item != NULL, RMT_ADDR_ERROR_STR, ESP_FAIL);
    RMT_CHECK(item_num > 0, RMT_DRIVER_LENGTH_ERROR_STR, ESP_ERR_INVALID_ARG);
    rmt_obj_t* p_rmt = p_rmt_obj[channel];
    int block_num = RMT.conf_ch[channel].conf0.mem_size;
    int item_block_len = block_num * RMT_MEM_ITEM_NUM;
    int item_sub_len = block_num * RMT_MEM_ITEM_NUM / 2;
    int len_rem = item_num;
	if (wait_tx_done) xSemaphoreTake(p_rmt->tx_sem, portMAX_DELAY);  // if added here
`
Using the code in the above example it is possible to stop RMT transmission when using looping. However it does not work well for short time delays. For example, if the three commands to stop transmission are issued 30ms after transmission is initiated, it will take substantially longer than 30 ms to stop.

The above two examples are using looping. If we send a fewer number of pulses, rmt_write_items will not use the ISR to send the pulses. This simplifies the operation of the RMT device. If rmt_tx_stop works anywhere, it should work here but it does not. 

Ideally the rmt_tx_stop code (and other code) in rmt.c should be modified so that transmission could be stopped at any time by issuing only the rmt_tx_stop command.

csann, 2018-02-02:
What is the status of this issue?

negativekelvin, 2018-02-02:
So the isr should probably check the value of tx_start and to stop faster you would have to set the duration of the items to zero or do some hack like changing the memory owner to rx.

csann, 2018-02-04:
@negativekelvin 

Thank you for your suggestion. I am trying to implement it without success. Here is what I have found:

To set the value of the tx_start bit, I use 
      RMT.conf_ch[1].conf1.tx_start = 0;

This command does nothing. The reason the rmt_tx_stop command does not work is because setting this bit to 0 does nothing. **This is the core problem I am having with the Remote peripheral. Setting the tx_start bit to 0 does nothing.** 

Next I create a rmt_items structure that contains only one item with level0=1, duration0=0, level1=0, and duration1=0.

Next I attempt to use 
     rmt_write_items(1, empty_rmt_item, 1, true); 
to send the structure to the rmt.  Calling this command within the ISR causes a panic. Apparently I cannot overwrite rmt_items while the rmt is transmitting. Or possibly the rmt_write_items command takes too long to execute in the ISR. 

Do you have any suggestions on how I can set the duration of the next unprocessed rmt_item to zero?

csann, 2018-02-05:
I found how to stop the rmt. 

for (int i = 0; i < 64; i++) {		
     RMTMEM.chan[(int) channel].data32[i].val = 0;
}

No need to set the tx_start bit since that doesn't accomplish anything. Nor is there any need to change ownership of the memory.

negativekelvin, 2018-02-05:
Use whatever works but changing ownership should be faster and generate an almost immediate interrupt and it is a technique used in RX mode already.

csann, 2018-02-05:
OK, I will try that. I'll also inspect rmt.c to see how it was used in RX mode. 

Thanks for the info!

koobest, 2018-02-06:
Hi, @csann，
    Yes，As you said. you can rewrite rx_tx_stop as follows:
`
esp_err_t rmt_tx_stop(rmt_channel_t channel)
{
    RMT_CHECK(channel < RMT_CHANNEL_MAX, RMT_CHANNEL_ERROR_STR, ESP_ERR_INVALID_ARG);
    portENTER_CRITICAL(&rmt_spinlock);
    RMTMEM.chan[channel].data32[0].val = 0;
    RMT.conf_ch[channel].conf1.tx_start = 0;
    RMT.conf_ch[channel].conf1.mem_rd_rst = 1;
    RMT.conf_ch[channel].conf1.mem_rd_rst = 0;
    portEXIT_CRITICAL(&rmt_spinlock);
    return ESP_OK;
}
`
we will update the rmt driver in new version soon. thanks!!.

csann, 2018-02-06:
Thank you @koobest for your information! I'm happy to see this problem is being addressed in the IDF.

I do have an additional question. What does tx_start and mem_rd_rst do? 
```
RMT.conf_ch[channel].conf1.tx_start = 0;
RMT.conf_ch[channel].conf1.mem_rd_rst = 1;
RMT.conf_ch[channel].conf1.mem_rd_rst = 0;
```
The TRM doesn't explain these bits very well. It basically says tx_start causes transmission to start and stop but according to my tests it doesn't do that. The TRM also says the rd_rst bit causes the memory address to be reset. I'm not sure what that means. 

Is it possible to modify the TRM to explain more about what happens when the tx_start and mem_rd_rst are toggled?

@negativekelvin is also correct. You can simply change the memory owner to cause transmissions to stop.

negativekelvin, 2018-02-06:
Basically this

```
RMTMEM.chan[channel].data32[0].val = 0;. //set item 0 to 0 (including duration)
RMT.conf_ch[channel].conf1.tx_start = 0; //clear start bit
RMT.conf_ch[channel].conf1.mem_rd_rst = 1;. //reset read pointer to item 0
RMT.conf_ch[channel].conf1.mem_rd_rst = 0;. //clear reset bit
```
The advantage is it doesn't generate an error interrupt but the disadvantage is it changes the item data and maybe is not absolutely as fast as change mem owner

csann, 2018-02-07:
@negativekelvin, @koobest 

Thank you for your comments. They really help. I better understand what those bits do. I still don't understand what the tx_start bit does since it seems to do nothing for me.

Unfortunately, there is a problem with the code. I tried putting the 4 lines @negativekelvin shows in my ISR and I also tried putting @koobest code in rmt_tx_stop (with the RMT_CHECK and spin lock code). In both cases I get repeated core panics.

```Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x40082b28  PS      : 0x00060034  A0      : 0x40082334  A1      : 0x3ffb0be0  
0x40082b28: rmt_driver_isr_default at /Users/clark/Desktop/esp32-workspace/MotorControl/main/./rmt.c:646

0x40082334: _xt_lowint1 at /Users/clark/Documents/esp/esp-idf/components/freertos/./xtensa_vectors.S:1105

A2      : 0x00000013  A3      : 0x00000008  A4      : 0x0000001f  A5      : 0x3ff56000  
A6      : 0x00000000  A7      : 0x80000001  A8      : 0x80082b65  A9      : 0x3ffb0bc0  
A10     : 0x00000000  A11     : 0x3ffb0be0  A12     : 0x3ffb4294  A13     : 0x00060021  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x400d1d4a  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082b28
0x400d1d4a: esp_vApplicationIdleHook at /Users/clark/Documents/esp/esp-idf/components/esp32/./freertos_hooks.c:85

0x40082b28: rmt_driver_isr_default at /Users/clark/Desktop/esp32-workspace/MotorControl/main/./rmt.c:646

Backtrace: 0x40082b28:0x3ffb0be0 0x40082331:0x3ffb0c10 0x400d1d47:0x00000000
0x40082b28: rmt_driver_isr_default at /Users/clark/Desktop/esp32-workspace/MotorControl/main/./rmt.c:646

0x40082331: _xt_lowint1 at /Users/clark/Documents/esp/esp-idf/components/freertos/./xtensa_vectors.S:1105

0x400d1d47: esp_vApplicationIdleHook at /Users/clark/Documents/esp/esp-idf/components/esp32/./freertos_hooks.c:85

Core 0 register dump:
PC      : 0x400d1d4a  PS      : 0x00060034  A0      : 0x8008576d  A1      : 0x3ffb4fd0  
0x400d1d4a: esp_vApplicationIdleHook at /Users/clark/Documents/esp/esp-idf/components/esp32/./freertos_hooks.c:85

A2      : 0x00000008  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffb55e8  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffb23e0  A9      : 0x3ffb23c4  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d1d4a:0x3ffb4fd0 0x4008576a:0x3ffb4ff0
0x400d1d4a: esp_vApplicationIdleHook at /Users/clark/Documents/esp/esp-idf/components/esp32/./freertos_hooks.c:85

0x4008576a: prvIdleTask at /Users/clark/Documents/esp/esp-idf/components/freertos/./tasks.c:4571

```

koobest, 2018-02-08:
Hi, @csann, did you change the ownership of the memory and whether callback are used in ISR? I'm not clear where you modified, can you provide more informations?

csann, 2018-02-08:
@koobest 

No I did not change the ownership of the memory and I used no callbacks in my ISR.

I performed three tests. The first test was using a GPIO interrupt to attempt to stop the RMT from transmitting. I used only part of your code. Here is the exact code I used in my ISR:

```
static void IRAM_ATTR abs_stepper_home_flag_found_isr_handler(void *channel) {
	RMTMEM.chan[(int) channel].data32[0].val = 0;
	RMT.conf_ch[(int) channel].conf1.tx_start = 0;
	RMT.conf_ch[(int) channel].conf1.mem_rd_rst = 1;
	RMT.conf_ch[(int) channel].conf1.mem_rd_rst = 0;
}
```
When I trigger this ISR to run, the RMT immediately stops transmitting, then it panics, reboots, and panics.

For the second test I put the code you posted inside the rmt_tx_stop function in rmt.c. Then I modified the ISR to call rmt_tx_stop as follows:

```
IN RMT.C:
esp_err_t rmt_tx_stop(rmt_channel_t channel) {
	ESP_LOGD(RMT_TAG, "Stopping in driver");
	RMT_CHECK(channel < RMT_CHANNEL_MAX, RMT_CHANNEL_ERROR_STR, ESP_ERR_INVALID_ARG);
	portENTER_CRITICAL(&rmt_spinlock);
	RMTMEM.chan[(int) channel].data32[0].val = 0;
	RMT.conf_ch[(int) channel].conf1.tx_start = 0;
	RMT.conf_ch[(int) channel].conf1.mem_rd_rst = 1;
	RMT.conf_ch[(int) channel].conf1.mem_rd_rst = 0;
	portEXIT_CRITICAL(&rmt_spinlock);
	return ESP_OK;
}

IN MY PROGRAM:
static void IRAM_ATTR abs_stepper_home_flag_found_isr_handler(void *channel) {
	rmt_tx_stop((int) channel);
}
```
When I trigger this ISR to run, the RMT immediately stops transmitting, then it panics, reboots, and panics, exactly as it did in the first test.

For the third test, I modified the ISR to change the owner of the RMT memory as follows:
```
static void IRAM_ATTR abs_stepper_home_flag_found_isr_handler(void *channel) {
	rmt_set_tx_loop_mode((int) channel, false);
	rmt_set_memory_owner((int) channel, RMT_MEM_OWNER_RX);
}
```
This code works fine. It stops the RMT from transmitting and does not cause panics.

It appears to me that the code you posted has a problem and will not run either in an ISR or in the rmt_tx_stop function.

I hope this helps. Please let me know if I need to provide more information.

koobest, 2018-02-08:
Hi @csann 
Please have a check on the gpio isr handle parameter，I think it should be  `*(int *)channel`  if not a typo.
thanks!!

koobest, 2018-02-08:
Hi, @csann 
I tried to call rmt_tx_stop in the interrupt, but did not crash.

csann, 2018-02-08:
@koobest 
I believe the gpio isr handler parameter type is void *. Here is the gpio isr handler from the ESP IDF gpio example (esp-idf/examples/peripherals/gpio/main/gpio_example_main.c):

```static` void IRAM_ATTR gpio_isr_handler(void* arg)```

I do not understand why my program crashes and yours does not. I am going to make a simple project consisting of a RMT that is configured to loop indefinitely and a gpio/isr which causes the RMT to stop transmitting. If it works, then I will figure out what is different with my code. If it doesn't work, I will post the complete project so you can see what I am doing.

csann, 2018-05-14:
I need it to stop immediately.

I am using the rmt to control a stepper. Suppose I command a stepper to move 30 steps at a certain speed. In this case I would create a rmt_item with 30 pulses with the correct duration. When I send the rmt_item the stepper will begin moving. If my needs change, I may have to abort the move and immediately begin another move, perhaps in the other direction. So I would need to immediately stop transmission of the current rmt_item. I would then issue a new rmt_item.

But suppose I command a stepper to move 3000 steps at a certain speed. This is perhaps too long to create a rmt_item with so many pulses. In this case I would create a rmt_item with 1 pulse of the correct duration and set the rmt to loop. Meanwhile I would use a counter to count the pulses. The counter would stop the rmt when 3000 pulses were sent. It would be good if the counter could use the tx_stop command to do this too.

I do not see why the tx_stop command is ever needed if someone wanted to stop when the rmt_item is finished sending. If I want it to stop when the current rmt_item finishes sending, I don’t need to issue the tx_stop command, I just need to wait and it will stop as soon as there is no more data. Perhaps I am missing something?

koobest, 2018-05-14:
@csann , 
    thanks for your feedback.  We will add this feature to the hardware in later versions of the chip.
we can also control the stepper motor via LEDC.  you can refer to the examples:
https://github.com/espressif/esp-iot-solution/blob/master/components/motor/stepper/a4988/a4988.cpp

Hope it helps.

closed 2018-05-15 by FayeY