espressif/esp-idf #1457: xTaskCreate -> Guru Meditation Error of type InstrFetchProhibited occurred on core  0.

Kazuya91, 2018-01-01:
I am trying to create Tasks and synchronize them with xTaskNotifyGive. When I create the Task it crashes with the Error given in the title.
What I personally dont understand is the fact that the code is from the example site except some printouts. Example Code from FreeRTOS: https://www.freertos.org/xTaskNotifyGive.html

My Code:

```
#include <string.h>
#include <time.h>
#include <sys/time.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_attr.h"
#include "esp_deep_sleep.h"
#include "nvs_flash.h"
#include <cJSON.h>

#include "lwip/err.h"
#include "apps/sntp/sntp.h"
#include <curl/curl.h>

/* FreeRTOS event group to signal when we are connected & ready to make a request */
static EventGroupHandle_t wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
const int CONNECTED_BIT = BIT0;

static const char *TAG = "example";

/* Variable holding number of times ESP32 restarted since first boot.
 * It is placed into RTC memory using RTC_DATA_ATTR and
 * maintains its value when ESP32 wakes from deep sleep.
 */
RTC_DATA_ATTR static int boot_count = 0;

static void prvTask1( void *pvParameters );
static void prvTask2( void *pvParameters );
static TaskHandle_t xTask1 = NULL, xTask2 = NULL;

static void prvTask1( void *pvParameters )
{
    for( ;; )
    {
        /* Send a notification to prvTask2(), bringing it out of the Blocked
        state. */
    		ESP_LOGI(TAG, "I am Task1, I notify Task2");
        xTaskNotifyGive( xTask2 );

        /* Block to wait for prvTask2() to notify this task. */
        ESP_LOGI(TAG, "I am Task2 and I wait for the notification of Task1");
        ulTaskNotifyTake( pdTRUE, portMAX_DELAY );
    }
}
/*-----------------------------------------------------------*/

static void prvTask2( void *pvParameters )
{
    for( ;; )
    {
        /* Block to wait for prvTask1() to notify this task. */
    		ESP_LOGI(TAG, "I am Task2 and I wait for the notification of Task1");
        ulTaskNotifyTake( pdTRUE, portMAX_DELAY );

        /* Send a notification to prvTask1(), bringing it out of the Blocked
        state. */
        ESP_LOGI(TAG, "I am Task2, I notify Task1");
        xTaskNotifyGive( xTask1 );

    }
}

void app_main(){

	if (boot_count == 0){

		//-----------------Boot-------------------------
		ESP_LOGI(TAG, "COLD START!");
		ESP_LOGI(TAG, "Boot count: %d", boot_count);
		boot_count++;

		//-----------------Task_Creation----------------

		xTaskCreate( prvTask1, "Task1", 200, NULL, tskIDLE_PRIORITY, &xTask1 );
		xTaskCreate( prvTask2, "Task2", 200, NULL, tskIDLE_PRIORITY, &xTask2 );
		vTaskStartScheduler();

		
	}
}
```

The Error-Print of "makemonitor":
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:5096
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:12664
entry 0x40078f50
I (34) boot: ESP-IDF v3.0-dev-203-gdce7fcb9-dirty 2nd stage bootloader
I (34) boot: compile time 14:29:56
I (139) boot: Enabling RNG early entropy source...
I (140) boot: SPI Speed      : 80MHz
I (140) boot: SPI Mode       : DIO
I (169) boot: SPI Flash Size : 4MB
I (207) boot: Partition Table:
I (241) boot: ## Label            Usage          Type ST Offset   Length
I (309) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (379) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (449) boot:  2 factory          factory app      00 00 00010000 00100000
I (519) boot: End of partition table
I (558) boot: Disabling RNG early entropy source...
I (609) boot: Loading app partition at offset 00010000
I (663) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x076dc ( 30428) map
I (783) esp_image: segment 1: paddr=0x00017704 vaddr=0x3ffb0000 size=0x02d88 ( 11656) load
I (843) esp_image: segment 2: paddr=0x0001a494 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at ??:?

I (911) esp_image: segment 3: paddr=0x0001a89c vaddr=0x40080400 size=0x05774 ( 22388) load
I (1025) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x26108 (155912) map
0x400d0018: _flash_cache_start at ??:?

I (1271) esp_image: segment 5: paddr=0x00046128 vaddr=0x40085b74 size=0x0d19c ( 53660) load
0x40085b74: prvProcessReceivedCommands at /Users/serhat/esp/esp-idf/components/freertos/./timers.c:660

I (1351) esp_image: segment 6: paddr=0x000532cc vaddr=0x400c0000 size=0x00000 (     0) load
I (1352) esp_image: segment 7: paddr=0x000532d4 vaddr=0x50000000 size=0x00004 (     4) load
I (1452) cpu_start: Pro cpu up.
I (1453) cpu_start: Starting app cpu, entry point is 0x40080e00
0x40080e00: call_start_cpu1 at /Users/serhat/esp/esp-idf/components/esp32/./cpu_start.c:192

I (1514) cpu_start: App cpu up.
I (1552) heap_init: Initializing. RAM available for dynamic allocation:
I (1616) heap_init: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1673) heap_init: At 3FFB66C8 len 00029938 (166 KiB): DRAM
I (1732) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1791) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1851) heap_init: At 40092D10 len 0000D2F0 (52 KiB): IRAM
I (1909) cpu_start: Pro cpu start user code
I (2065) cpu_start: Starting scheduler on PRO CPU.
I (2068) cpu_start: Starting scheduler on APP CPU.
I (2068) example: COLD START!
I (2068) example: Boot count: 0
Guru Meditation Error of type InstrFetchProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x800e3a38  PS      : 0x00050033  A0      : 0x800e3a38  A1      : 0x3ffb75c0  
A2      : 0x3ffb4e64  A3      : 0x3f404f7c  A4      : 0x50000000  A5      : 0x3ffb7610  
A6      : 0x3ffb75f0  A7      : 0x0000000c  A8      : 0x00000000  A9      : 0x800d07aa  
A10     : 0x3ffb7630  A11     : 0x7fffffff  A12     : 0x50000000  A13     : 0x00000001  
A14     : 0x3ffb5230  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000014  
EXCVADDR: 0x800e3a38  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400e3a38:0x3ffb75c0 0x400e3a35:0x3ffb75f0
0x400e3a38: app_main at /Users/serhat/esp/esp-idf/examples/get-started/hello_world/main/./hello_world_main.c:101

0x400e3a35: app_main at /Users/serhat/esp/esp-idf/examples/get-started/hello_world/main/./hello_world_main.c:90 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

mahavirj, 2018-01-01:
I would recommend few changes to test code as below:
1. vTaskStartScheduler() this is not required, as scheduler is already started before app_main() call
2. Stack sizes for both prvTask1 and prvTask2 are too suboptimal, trying increasing them to few KB

chegewara, 2018-01-01:
```
		xTaskCreate( prvTask1, "Task1", 200, NULL, tskIDLE_PRIORITY, &xTask1 );
		xTaskCreate( prvTask2, "Task2", 200, NULL, tskIDLE_PRIORITY, &xTask2 );
```
Here is the issue. Esp32 tasks require minimum about 760kB and you have set only 200. For tests you can use 4kB or more.

Kazuya91, 2018-01-01:
Thank you guys! Its working. But I have another issue now.... I try to send JSON data via curl. I have a local webserver with the django-framework. If my server receives something, it gives me a printout which it does in this case. So my ESP32 is sending data and my server receives it. The problem is that it crashes with an Error called:

Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.

my Code looks like this right now:

```
int simplecurl(){

	  CURL *curl;
	  CURLcode res;

	  static const char *postthis = "{ \"mac\" : \"24:0a:c4:04:fd:28\" , \"battery\" : \"100\" , \"message\" : \"You got a message\"}";

	  curl = curl_easy_init();
	  if(curl) {
	    curl_easy_setopt(curl, CURLOPT_URL, "http://192.168.178.34:8000/kalender/returnjson/");
	    curl_easy_setopt(curl, CURLOPT_POSTFIELDS, postthis);

	    /* if we don't provide POSTFIELDSIZE, libcurl will strlen() by
	       itself */
	    curl_easy_setopt(curl, CURLOPT_POSTFIELDSIZE, (long)strlen(postthis));

	    /* Perform the request, res will get the return code */
	    res = curl_easy_perform(curl);
	    /* Check for errors */
	    if(res != CURLE_OK)
	      fprintf(stderr, "curl_easy_perform() failed: %s\n",
	              curl_easy_strerror(res));

	    /* always cleanup */
	    curl_easy_cleanup(curl);
	  }
	  return 0;

}

void app_main(){

	//first boot
	//do everything here just once
	if (boot_count == 0){

		//-----------------Boot-------------------------
		ESP_LOGI(TAG, "COLD START!");
		ESP_LOGI(TAG, "Boot count: %d", boot_count);
		boot_count++;

		//------------------WIFI-----------------------

		initialise_wifi();
		xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, true, portMAX_DELAY);
		printf("\nI am Connected!!\n");

		//------------------Curl-----------------------
		xTaskCreate(&simplecurl, "Curl", 10240, NULL, 5, NULL);

}
```
I erased all other code so i can be sure its really the lines of code above.

This is the Error-Log.

```
I (5067) wifi: connected with SSID-censored, channel 11
I (6017) event: ip: 192.168.178.31, mask: 255.255.255.0, gw: 192.168.178.1

I am Connected!!
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40100618  PS      : 0x00060f30  A0      : 0x00000000  A1      : 0x3ffc5f80  
0x40100618: simplecurl at /Users/serhat/esp/esp-idf/examples/get-started/hello_world/main/./hello_world_main.c:164

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80100616  A9      : 0x3ffc5f60  
A10     : 0x3ffc62bc  A11     : 0x0000003c  A12     : 0x00000052  A13     : 0x3ffc600c  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x40100618:0x3ffc5f80 0x7ffffffd:0x3ffc5fa0
0x40100618: simplecurl at /Users/serhat/esp/esp-idf/examples/get-started/hello_world/main/./hello_world_main.c:164

Rebooting...
ets Jun  8 2016 00:22:57
```

I really have no clue.. I wish the Error messages would be more detailed or precise...

I appreciate your help! Thanks a lot

ginkgm, 2018-01-02:
strange pointer..try
```
void simplecurl(void*);
xTaskCreate(simplecurl, "Curl", 10240, NULL, 5, NULL);
```

ginkgm, 2018-01-03:
BTW, FreeRTOS tasks should not return, whether
1. loop infinitely in the task by ``while(1) { ... }``
or
2. put a ``vTaskDelete(NULL);`` at the end of the task function to kill itself to avoid such crash.

Kazuya91, 2018-01-03:
@ginkgm 
Ty very much. It wasnt the pointer. As you said, i need to delete the Task when its finished. A vTaskDelete("Handle of the Task") at the End of the function definition was the line of code I needed. I didnt know that Tasks are periodic in FreeRTOS.

Do you know a way of creating Tasks and keep them even after deepsleep? My Goal is to minimize the time the ESP32 is awake. I want to create the Task just if its a "cold start" (first boot) of the ESP32 and after deepsleep he shall repeat task and so on. In my opinion it doesnt make any sense to create and delete it every time the ESP32 wakes up from deep sleep.

chegewara, 2018-01-03:
Task does not have to be deleted, just cant exit. You can have endless loop in such task.

projectgus, 2018-01-04:
I'm going to close this as it seems like it's resolved.

For future discussions about FreeRTOS programming (that aren't related to specific bugs in ESP-IDF), the forum is a good place to post: https://esp32.com

closed 2018-01-04 by projectgus

abanejad, 2019-07-20:
Had the same issue

abanejad, 2019-07-20:
Had the same issue
had to search a bit a change my code till i I finally found out that calling `vTaskStartScheduler() ` is not needed since is running when esp-idf calls the main function. I Deleted it and it started working. If this  is true please correct the API references for freertos so we wont be misled.
Thank You

chegewara, 2019-07-20:
@abanejad Could you paste link to API reference you its misleading?

abanejad, 2019-07-20:
@chegewara yeah sure [Espressif's esp-idf's official documentation on Freertos](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/freertos.html)
As you can see there is an example and few other macros refering to the `vTaskStartScheduler()`
Based on this I was struggling a while because I didn't know the fact I just mentioned before and I found it out after searching it though other websites. So if it's correct I guess the documentation should be updated

chegewara, 2019-07-20:
@projectgus @krzychb

projectgus, 2019-07-22:
Hi @abanejad ,

Thanks for pointing this out. That docs page is generated from the FreeRTOS docs directly, and you're right that this difference (scheduler already running when app_main() runs), is not indicated there. Will fix.