espressif/esp-idf #1459: AP Disappear

Pratikhyadav, 2018-01-03:
Hello 

I created Simple Rest Server using netcon library.
Simple Rest Server Works Fine.
But When I introduce Flash Write Operation.
I am Writing to flash from 0x300000 location.
Size of Data Written into flash is 900KB.
My AP disappear After 3 Hour.

My partition file detail is As following:

Name, Type, SubType, Offset, Size, Flags
nvs, data, nvs, 0x9000, 0x4000,
otadata, data, ota, 0xd000, 0x2000,
phy_init, data, phy, 0xf000, 0x1000,
factory, app, factory, 0x10000, 0x100000,
ota_0, app, 0x10, 0x110000, 0x100000,
storage, data, 0x82, 0x210000, 0xC0000,

When I Replaced Flash Write Operation with 100 ms Delay, Which works fine more than 12 hours. And AP Not Disappear.

Is there Anything I missed.?

Let me Know If anyone requires More Detail about it.

jack0c, 2018-01-14:
Duplicated with #1372 @FayeY

FayeY, 2018-01-15:
Closing this, as this issue is duplicated with #1372.

closed 2018-01-15 by FayeY