espressif/esp-idf #1460: recvmsg / IP_PKTINFO support - upgrade LwIP?

SStorm, 2018-01-03:
The later versions of LwIP implement support for recvmsg using IP_PKTINFO, which is very useful when you want to get the actual destination address for received UDP packets. Are there any plans to upgrade the LwIP impl that's bundled with esp-idf at this stage, or should I be looking for alternative solutions? Thanks!

TimXia, 2018-01-04:
Yes, we are planing to synchronize the latest official LwIP with that in esp-idf. Maybe in the next version.

SStorm, 2018-01-04:
Are there any milestones for this, i.e. when could this be reasonably expected? i.e. is this 1-2 months or 6+ months away? Thanks!

TimXia, 2018-01-05:
Maybe 3~4 months.

liuzfesp, 2018-01-22:
@SStorm LWIP will update to latest version in v3.1 @zhangyanjiaoesp

SStorm, 2018-04-10:
@zhangyanjiaoesp There's a lot of usages in https://github.com/yarrick/lwip/, i.e. https://github.com/yarrick/lwip/blob/5ee77262bc6003c869b77ea606a4d80b2285d049/src/api/sockets.c

zhangyanjiaoesp, 2018-04-11:
I will update the lwip master and check it , thanks.

SStorm, 2018-07-25:
Hi guys, just wanted to check if there's any updated on this? I'm guessing my PR isn't going to be accepted at all?..

TimXia, 2018-07-26:
@SStorm We are working on it. It will be done in IDF v3.2.

zhangyanjiaoesp, 2018-09-07:
@SStorm We have update the lwIP in the lastest master, and add the function [lwip_recvmsg](https://github.com/espressif/esp-lwip/blob/18548d1e256f7651fb4da48dbd770556e0eca8ce/src/api/sockets.c#L1256) . You can test if it's useful for you. 
Sorry for replying so later.

SStorm, 2018-09-07:
@zhangyanjiaoesp thanks! Will try it out as soon as I have a moment.

projectgus, 2019-02-17:
Reopening as the support for recvmsg() is in ESP-IDF master but we don't have support for IP_PKTINFO yet.

ESP-IDF currently uses a [version of LWIP based on v2.0.3](https://github.com/espressif/esp-lwip) in master. Support for PKTINFO was added to LWIP in commit [2f117add7](http://git.savannah.nongnu.org/cgit/lwip.git/commit/?id=2f117add7) and released in LWIP v2.1.0.

Unfortunately, ESP-IDF will probably not add support for IP_PKTINFO until after we update ESP-LWIP to LWIP v2.1.0 or newer.

Alvin1Zhang, 2019-03-15:
@projectgus Thanks for the update.

freakyxue, 2019-11-13:
hi @SStorm 
LwIP 2.1.2 has added this function, we will discuss whether to add this function in LwIP 2.0.3

freakyxue, 2020-03-09:
hi @SStorm
 LwIP 2.1.2 has added this function，LwIP version of release 4.0 has been upgraded to 2.1.2

Alvin1Zhang, 2020-03-09:
@SStorm Thanks for reporting. This feature has been supported in ESP-IDF v4.0. If you any other feature/issue request, please help file another ticket. Thanks.

closed 2020-03-09 by Alvin1Zhang