espressif/esp-idf #1463: Fixed PIN-Code for Bluetooth

gadthrawn, 2018-01-03:
Hello,

Back with bluetooth classic in old days it was possible to set a fixed PIN-Code which you need to enter on other devices connecting to a bluetooth device (e.g. HC05).

Normally we used this during configuration for devices which do not have any display.

In the examples for Bluetooth LE i do not find any way to just say: Use this passphrase/PINcode, only some autogenerated things which is really not what i need.

Is there any possibility to say: 

ESP32 use this PINCode and ask for it if anyone tries to connect?

Maybe with Bluetooth Classic?

chegewara, 2018-01-03:
Here is answer on your question:

https://github.com/espressif/esp-idf/issues/1442

gadthrawn, 2018-01-03:
ah.. that means the answer is "not possible" at the moment.

chegewara, 2018-01-03:
It can be secure by using encryption but doesnt provide authentication if esp32 does not have display and/or numeric keypad.

gadthrawn, 2018-01-03:
Cryptography is not security. Going back to STM32 and HM-10 for the moment.

chegewara, 2018-01-03:
If you will connect it with boundind (which works) and whitelist devices connection and/or scan whitelist allowed (also works) you can have basic security level

closed 2018-01-08 by FayeY