espressif/esp-idf #1467: Guru Meditation Error

samimd19, 2018-01-05:
I am using an ESP32 to send temperature and Humidity sensor data to a MQQT broker. I am using the below mentioned sketch and it is showing the Guru Meditation Error and rebooting. As a beginner, I am finding it hard to debug the error. The error is also mentioned under the sketch.

Arduino sketch:

#include <WiFi.h>
#include <PubSubClient.h>
#include <SimpleDHT.h>

#define DHTPIN 2 // what pin we're connected to
int lamp = 15; 
const int LED_PIN = 5;

int pinDHT11 = 14;
SimpleDHT11 dht11;

// WiFi network name and password:
const char * networkName = "router-wifi-2048";
const char * networkPswd = "Saint_12345";

const char* mqtt_server = "192.168.1.5";
const char* mqttUser= "Raspbi";
const char* mqttPassword= "password";

WiFiClient espClient;
PubSubClient client(espClient);

void connectToWiFi(const char * ssid, const char * pwd)
{
  int ledState = 0;

  printLine();
  Serial.println("Connecting to WiFi network: " + String(ssid));

  WiFi.begin(ssid, pwd);

  while (WiFi.status() != WL_CONNECTED) 
  {
    // Blink LED while we're connecting:
    digitalWrite(LED_PIN, ledState);
    ledState = (ledState + 1) % 2; // Flip ledState
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.println("WiFi connected!");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void printLine()
{
  Serial.println();
  for (int i=0; i<30; i++)
    Serial.print("-");
  Serial.println();
}

void callback(char* topic, byte* message, unsigned int length) {
  //Serial.print("Message arrived on topic: ");
  //Serial.print(topic);
  //Serial.print(". Message: ");
  String messageTemp;
  
  for (int i = 0; i < length; i++) {
    //Serial.print((char)message[i]);
    messageTemp += (char)message[i];
  }
  //Serial.println();

  // Feel free to add more if statements to control more GPIOs with MQTT

  // If a message is received on the topic room/lamp, you check if the message is either on or off. Turns the lamp GPIO according to the message
  if(topic=="room/lamp"){
      Serial.print("Changing Room lamp to ");
      if(messageTemp == "on"){
        digitalWrite(lamp, HIGH);
        Serial.print("On");
      }
      else if(messageTemp == "off"){
        digitalWrite(lamp, LOW);
        Serial.print("Off");
      }
  }
  Serial.println();
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
   
    if (client.connect("ESP32Client", mqttUser, mqttPassword)) {
      Serial.println("connected");  
      // Subscribe or resubscribe to a topic
      client.subscribe("room/lamp");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup() {
  pinMode(lamp, OUTPUT);
  Serial.begin(115200);
  Serial.println("DHT test");
  connectToWiFi(networkName, networkPswd);;
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
  

}

void loop() {

if (!client.connected()) {
    reconnect();
  }
  if(!client.loop())
    delay(100);    
    client.connect("ESP32Client", mqttUser, mqttPassword);

   
  // read with raw sample data.
  byte temperature = 0;
  byte humidity = 0;
  byte data[40] = {0};
  if (dht11.read(pinDHT11, &temperature, &humidity, data)) {
    Serial.print("Read DHT11 failed");
    return;
  }
  Serial.println("");
  
  //Serial.print("Sample OK: ");

  client.publish("room/temperature",(const char*)temperature);
  client.publish("room/humidity", (const char*)humidity);
  Serial.print((int)temperature); Serial.print(" *C, ");
  Serial.print((int)humidity); Serial.println(" %");
  
  
  // DHT11 sampling rate is 1HZ.
  delay(1000);
}

Error shown:

Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400014e8  PS      : 0x00060830  A0      : 0x800d2693  A1      : 0x3ffd0e60  
A2      : 0x00000016  A3      : 0x00000014  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d2cc0  A9      : 0x3ffd0e30  
A10     : 0x00000002  A11     : 0x3f401153  A12     : 0x00000000  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000014  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014e8:0x3ffd0e60 0x400d2690:0x3ffd0e70 0x400d16bf:0x3ffd0e90 0x4010d808:0x3ffd0ee0

Rebooting...

chegewara, 2018-01-05:
Try this tool to debug errors in arduino-ide:
https://github.com/me-no-dev/EspExceptionDecoder

igrr, 2018-01-08:
Hi @samimd19, the crash looks like it is caused by access to a field of a structure/class, pointed to by a NULL pointer. You need to follow @chegewara's suggestion to get more information about the source of the issue.

closed 2018-01-23 by FayeY