espressif/esp-idf #1470: BLE security documentation (tutorial)

chegewara, 2018-01-05:
I want to say Thank You for this amazing documentation. Its very informative and really helpful. If i may suggest two changes:
- this event is also triggered when keys exchange is failed with fail_reason
> When the keys are exchanged successfully, the pairing process is done. This triggers an ESP_GAP_BLE_AUTH_CMPL_EVT event, which is used to print information such as remote device, address type and pair status

- please add info about ESP_GAP_BLE_NC_REQ_EVT event which is triggered when we have number comparison authentication (both peers set up ESP_IO_CAP_IO capability):
```
 case ESP_GAP_BLE_NC_REQ_EVT:
	 ESP_LOGI(LOG_TAG, "ESP_GAP_BLE_NC_REQ_EVT");
        ///the app will receive this evt when the IO  has DisplayYesNO capability and the peer device IO also has DisplayYesNo capability.
       ///show the passkey number to the user to confirm it with the number displayed by peer deivce.
       ESP_LOGE(GATTS_TABLE_TAG, "The passkey Notify number:%d", param->ble_security.key_notif.passkey);
     break;
```

Weijian-Espressif, 2018-03-14:
@chegewara thanks for your reminder, i will do this this week, thanks.

Weijian-Espressif, 2018-03-30:
thanks, the commit id is fc5b15173e9c67262492dca375f73b169f8e904c

closed 2018-03-30 by chegewara