espressif/esp-idf #1471: gatt_security_server is not working with PUBLIC address advertised

chegewara, 2018-01-06:
In this issue https://github.com/nkolban/esp32-snippets/issues/346 user reported problems with gatt_security_server which ive been tested and thats what ive found:
- after changing advertising address to PUBLIC and local_privacy to false of course application behave strange
- after reboot application can connect and pair/bond with android device (nRF connect app), ESP_OK
- when we have bonded esp32 with android and android phone is whitelisted we perform esp32 reboot and now we have few issues:
-- as we can see from this log (before esp32 reboot) android device is paired and bonded:
```
I (123422) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PENC
V (123422) SEC_GATTS_DEMO: GAP_EVT, event 9

I (123432) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PID
V (123432) SEC_GATTS_DEMO: GAP_EVT, event 8

I (123442) SEC_GATTS_DEMO: remote BD_ADDR: 781fdbc0c7f3
I (123442) SEC_GATTS_DEMO: address type = 0
I (123452) SEC_GATTS_DEMO: pair status = success
I (123452) SEC_GATTS_DEMO: Bonded devices number : 1

I (123462) SEC_GATTS_DEMO: Bonded devices list : 1

I (123462) SEC_GATTS_DEMO: 1c 71 fb 3f 02 00
V (126322) SEC_GATTS_DEMO: event = f

```
-- now we have logs after rebooting, im not sure but this may be the issue `E (13982) BT: L2CA_RemoveFixedChnl()  CID: 0x0006  BDA: 781fdbc0c7f3 not connected` because after this operation android device is removed from whitelist and normal connection cant be performed, also my phone shows me that esp32 is no longer bonded:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4328
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:11928
entry 0x40078dbc
I (150) cpu_start: Pro cpu up.
I (150) cpu_start: Starting app cpu, entry point is 0x40080f04
I (0) cpu_start: App cpu up.
I (153) heap_init: Initializing. RAM available for dynamic allocation:
D (160) heap_init: New heap initialised at 0x3ffaff10
I (165) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
D (171) heap_init: New heap initialised at 0x3ffca7f8
I (176) heap_init: At 3FFCA7F8 len 00015808 (86 KiB): DRAM
I (183) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (189) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (195) heap_init: New heap initialised at 0x400927f4
I (200) heap_init: At 400927F4 len 0000D80C (54 KiB): IRAM
I (207) cpu_start: Pro cpu start user code
D (219) clk: RTC_SLOW_CLK calibration value: 3242214
V (4) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (4) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (9) intr_alloc: Connected src 46 to int 2 (cpu 0)
V (14) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (20) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xC0E
D (28) intr_alloc: Connected src 57 to int 3 (cpu 0)
V (33) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (39) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (48) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (53) cpu_start: Starting scheduler on PRO CPU.
V (0) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (0) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E
D (10) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (10) cpu_start: Starting scheduler on APP CPU.
D (102) heap_init: New heap initialised at 0x3ffe0440
D (102) heap_init: New heap initialised at 0x3ffe4350
V (112) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (112) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (112) intr_alloc: Connected src 16 to int 12 (cpu 0)
D (132) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
D (152) BTDM_INIT: Release DRAM [0x3ffb2730] - [0x3ffb8000]

D (152) BTDM_INIT: Release DRAM [0x3ffbbb28] - [0x3ffbdb28]

D (152) BTDM_INIT: .data initialise [0x3ffae6e0] <== [0x4000d890]

D (162) BTDM_INIT: .bss initialise [0x3ffb0000] - [0x3ffb09a8]

D (172) BTDM_INIT: .bss initialise [0x3ffb09a8] - [0x3ffb1ddc]

D (172) BTDM_INIT: .bss initialise [0x3ffb1ddc] - [0x3ffb2730]

D (182) BTDM_INIT: .bss initialise [0x3ffb8000] - [0x3ffbbb28]

D (192) BTDM_INIT: .bss initialise [0x3ffbdb28] - [0x3ffc0000]

I (192) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (442) phy_init: loading PHY init data from application binary
D (442) nvs: nvs_open_from_partition phy 0
D (442) nvs: nvs_get cal_version 4
V (442) phy_init: phy_get_rf_cal_version: 366

D (452) nvs: nvs_get_str_or_blob cal_mac
D (452) nvs: nvs_get_str_or_blob cal_data
D (452) nvs: nvs_close 1
V (452) phy_init: register_chipv7_phy, init_data=0x3f427ff3, cal_data=0x3ffbcbb4, mode=0
I (482) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 0, 0
I (482) SEC_GATTS_DEMO: app_main init bluetooth
D (482) nvs: nvs_open_from_partition bt_config.conf 1
D (482) nvs: nvs_get_str_or_blob bt_cfg_key
D (492) nvs: nvs_close 2
D (492) nvs: nvs_open_from_partition bt_config.conf 1
D (492) nvs: nvs_set_blob bt_cfg_key 594
D (502) nvs: nvs_close 3
V (522) SEC_GATTS_DEMO: event = 0

V (532) SEC_GATTS_DEMO: event = 16

I (532) SEC_GATTS_DEMO: The number handle = 8
V (532) SEC_GATTS_DEMO: GAP_EVT, event 22

V (532) SEC_GATTS_DEMO: event = c

V (542) SEC_GATTS_DEMO: GAP_EVT, event 0

V (542) SEC_GATTS_DEMO: GAP_EVT, event 1

V (552) SEC_GATTS_DEMO: GAP_EVT, event 6

I (552) SEC_GATTS_DEMO: advertising start success
V (13642) SEC_GATTS_DEMO: event = e

I (13642) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
E (13982) BT: L2CA_RemoveFixedChnl()  CID: 0x0006  BDA: 781fdbc0c7f3 not connected
E (13982) BT: Device not found

E (13992) BT: BTA got unregistered event id 31

E (13992) BT: BTA got unregistered event id 31

V (14002) SEC_GATTS_DEMO: event = f

I (14002) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
D (14012) nvs: nvs_open_from_partition bt_config.conf 1
D (14012) nvs: nvs_set_blob bt_cfg_key 284
D (14072) nvs: nvs_close 4
V (14072) SEC_GATTS_DEMO: GAP_EVT, event 8

I (14072) SEC_GATTS_DEMO: remote BD_ADDR: 781fdbc0c7f3
I (14072) SEC_GATTS_DEMO: address type = 0
I (14072) SEC_GATTS_DEMO: pair status = fail
I (14082) SEC_GATTS_DEMO: Bonded devices number : 0

I (14082) SEC_GATTS_DEMO: Bonded devices list : 0

V (14092) SEC_GATTS_DEMO: GAP_EVT, event 6

I (14092) SEC_GATTS_DEMO: advertising start success
V (15652) SEC_GATTS_DEMO: GAP_EVT, event 20

V (27592) SEC_GATTS_DEMO: event = e

I (27592) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
V (29402) SEC_GATTS_DEMO: GAP_EVT, event 20

D (57932) nvs: nvs_open_from_partition bt_config.conf 1
D (57932) nvs: nvs_set_blob bt_cfg_key 284
D (57932) nvs: nvs_close 5
V (57932) SEC_GATTS_DEMO: GAP_EVT, event 8

I (57932) SEC_GATTS_DEMO: remote BD_ADDR: 781fdbc0c7f3
I (57942) SEC_GATTS_DEMO: address type = 0
I (57952) SEC_GATTS_DEMO: pair status = fail
I (57952) SEC_GATTS_DEMO: Bonded devices number : 0

I (57962) SEC_GATTS_DEMO: Bonded devices list : 0

E (57982) BT: btm_sec_clr_temp_auth_service() - no dev CB

E (57992) BT: Device not found

E (57992) BT: BTA got unregistered event id 31

E (57992) BT: BTA got unregistered event id 31

V (57992) SEC_GATTS_DEMO: event = f

I (58002) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
V (58012) SEC_GATTS_DEMO: GAP_EVT, event 6

I (58012) SEC_GATTS_DEMO: advertising start success

```

chegewara, 2018-01-06:
User just reported that this issue doesnt occur with android 7.1.1. My device is samsung S4 with android 5.0 and im having issues.

Weijian-Espressif, 2018-03-14:
@chegewara  we fix some bugs last year. Can you update to the latest IDF and try?

closed 2018-03-22 by FayeY

chegewara, 2018-03-26:
@Weijian-Espressif Updated, tested and seems to work even with my bluetooth v4.0 smartphone.