espressif/esp-idf #1472: Access to the Wifi firmware version

ammaree, 2018-01-06:
(307) wifi: wifi firmware version: 403db1d

#1 We have been looking to find the Wifi version as displayed at boot time, but cannot locate a macro or function anywhere. We would like to log the wifi firmware version to our cloud application to assist with diagnostics, configuration management and firmware upgrade decisions.

#2 Another thought would be if the option could be provided to decouple the wifi firmware from the application firmware. This way only a single copy of the wifi firmware could be stored on a partition of its own, thus making the application (Factory and OTA) images significantly smaller and the process faster.

Any comment and/or input on the above appreciated

igrr, 2018-01-06:
Option 2 is not possible, since the wifi driver is linked into the application. We have some ideas about loading wifi driver dynamically, but it is unlikely that this will be implemented soon.

Regarding getting the version of wifi driver, you can do so as part of the build process, in a way similar to the way IDF_VER variable is evaluated in the makefile. The version is simply the commit ID of the esp32-wifi-lib submodule.

ammaree, 2018-01-06:
Hi @igrr 
Thanks for the feedback.

So to, make sure I understand you correctly, I would need to do the equivalent shell command in my application make file and then add the symbol with -Dxxxx to the CFLAGS.
 
Last question, what would be the best (most correct/portable) syntax to refer to the "esp32-wifi-lib" working from an MSYS2 environment?

igrr, 2018-01-08:
Correct, although adding to CPPFLAGS would be more in line with the standard practice (CPPFLAGS are preprocessor flags, and are applied both to C and C++ source files).

You can get the short commit ID of the submodule like this:
```
git -C $IDF_PATH/components/esp32/lib rev-parse --short=8 HEAD
```

ammaree, 2018-01-11:
Thanks for the feedback but I cannot seem to get the same version info as is displayed when booting.

Using your command as below I get the following:

git -C ${IDF_PATH}/components/esp32/lib rev-parse --short=8 HEAD    ==>    "c19d8c7a"
git -C ${IDF_PATH}/components/esp32/lib rev-parse HEAD    ==>    "c19d8c7a6c8aa5884612c691262b6125327e0bca"

whereas the boot process shows        ==>    "403db1d"

Am I missing something?

igrr, 2018-01-11:
Sorry, i didn't understand that you were trying to get the version printed in the wifi driver initialization message, rather than the version of the wifi libraries submodule. Got it now.

I think the only way to get *that* version right now is something like
```
xtensa-esp32-elf-strings ${IDF_PATH}/components/esp32/lib/libnet80211.a | egrep '^[a-f0-9]{7}$'
```

If you can explain the use case for getting that version number (and not the esp32-wifi-lib submodule commit id), then I'm sure we will consider adding an API to obtain the version.

ammaree, 2018-01-11:
Hi @igrr ,

It actually does not matter. We just need a unique token that can be directly linked to the specific version of Wifi firmware running on a mote. Our choice to use the version printed during drive initialization was made because that is what you are using.

Any reason why you chose the version  number being displayed at initialization, should we not use the same version for consistency and ease of cross verification?

igrr, 2018-01-12:
I would say that knowing IDF version used to compile the application is sufficient for the purposes of reporting issues.
I don't fully understand what you mean by "cross verification" in this context. Do you mean that you will need to compare the version string printed by WiFi library with the version string determined by some other means?

liuzfesp, 2018-01-16:
Agree with @igrr , for application the IDF version is enough for reporting issues. We need to care about the WiFi firmware version only you use the IDF + a private WiFi firmware version.

ammaree, 2018-01-16:
Hi @igrr 

by cross verification we intended visual confirmation of the version shown when booting compared to the version being logged on the cloud.

Can I maybe suggest that the version shown when booting is selected to be the most relevant, and that a mechanism is available through the SDK to access this same version and use it for whatever purposes.

Having 2 different versions with the version displayed by the IDF not accessible through the SDK seems a bit complex or inconsistent?

closed 2018-01-16 by ammaree