espressif/esp-idf #1473: Weird GPIO behavior on WROVER-KIT DEV board

adherent88, 2018-01-07:
Hi @ all,

I observed some weird behaviors on wrover-kit dev board, which drives me crazy :(

**Behavior 1:** GPIO5's initial state is always 0(low).
 I configured this pin as input and enabled pull-up. So by theory I should get a 1(high) when I read it in application using **gpio_get_level(5)**. However it is the opposite. 

**Behavior 2:** GPIO16, GPIO17 are not affected by external input. 
These two pins are configured as inputs and pull-up is enabled. Falling edge and rising edge are registered to trigger interrupt. However when I apply a low voltage(~0.01V as I read from the multimeter) to either of them, the value I read in application using **gpio_get_level(16)** is all the time 1(high)! And also no interrupt is triggered. 

After I change **GPIO16 to GPIO4** which is used in IDF GPIO example, it works fine and as expected. 

I am totally lost here, can someone please help?

With many thanks!

llewellynnu, 2018-01-07:
Which module are u using ?

Llewellyn Fernandes
Lead - Hardware

adherent88, 2018-01-07:
hmmmmmm sorry, what do mean by which module? 

If you mean the hardware I worked on, it is as stated the ESP-WROVER-KIT itself.

llewellynnu, 2018-01-07:
Hi I meant are u using esp wroom module or esp wrover

Llewellyn Fernandes
Lead - Hardware

adherent88, 2018-01-07:
Hi Llewellyn 
I am using esp wrover. 
Actually I checked e.g. if GPIO16, 17 are occupied by some peripherals. But it seems to be not the case. They should be free.

llewellynnu, 2018-01-07:
Hi if m not mistaken gpio 16 and 17 are used for clk and cs of the psram
for the esp wrover module

Llewellyn Fernandes
Lead - Hardware

adherent88, 2018-01-07:
Hi Llewellyn,
u r right, I missed this:
[image]

my bad, thank you!
Btw, any explaination about behaviour 1?

negativekelvin, 2018-01-07:
7 | GPIO5 | Backlight
-- | -- | --

adherent88, 2018-01-07:
Thanks to all!

closed 2018-01-07 by adherent88